OctoPrint/OctoPrint #4634: DNS issue causes unresponsiveness

crysxd, 2022-08-28:
### The problem

I have a curious issue on one of my OctoPrints with DNS. I'm not sure what the exact cause is, but my expectation when OctoPrint can't reach the internet is that it's core functionality stays usable. Unfortunately, the page takes forever to load and once it's loaded commands, e.g. turning on the PSU, are not executed. The HTTP requests take forever to complete.

I was digging through the logs and found the the software update and plugin manager seem to constantly try to make requests, somehow blocking the thread responsible for handling incoming requests.

Below is from `octoprint.log`. I reload the page in the browser around `12:01:28`, the below logs show that OctoPrint is "busy" with software update and plugin manger stuff (showing my DNS issue) but only at `12:01:54` the user log out (from the old page being closed) and a new log in (from the new page being loaded) is shown. This also corresponds with the browser showing the page at that exact time.
 
``` 
2022-08-28 12:01:34,144 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-08-28 12:01:34,283 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2022-08-28 12:01:36,756 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2022-08-28 12:01:44,370 - octoprint.plugins.softwareupdate - WARNING - Could not check firmware_check for updates due to a network error
2022-08-28 12:01:49,354 - octoprint.plugins.softwareupdate - WARNING - Could not check pip for updates due to a network error
2022-08-28 12:01:54,359 - octoprint.plugins.softwareupdate - WARNING - Could not check file_check for updates due to a network error
2022-08-28 12:01:54,422 - octoprint.plugins.softwareupdate - WARNING - Could not check pi_support for updates due to a network error
2022-08-28 12:01:54,568 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2022-08-28 12:01:54,743 - octoprint.server.util.flask - INFO - Passively logging in user admin from fe80::c27:6af4:ced1:316a
2022-08-28 12:01:54,743 - octoprint.access.users - INFO - Cleaning up user session 16C82ADB764449FCA54944B1A08A0D43 for user admin
2022-08-28 12:01:54,744 - octoprint.server.util.sockjs - INFO - User admin logged out, logging out on socket
2022-08-28 12:01:54,747 - octoprint.access.users - INFO - Logged out user: admin
2022-08-28 12:01:54,749 - octoprint.access.users - INFO - Logged in user: admin
```

Currently some dev version is installed, but due to the lack of internet I can't verify the issue also surfaces in 1.8. Ping to the OctoPi is constantly below 2ms with an Ethernet 1GBit connection

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

2.0.0.dev1552+g587e73262 

### Operating system running OctoPrint

OctPi

### Printer model & used firmware incl. version

/

### Browser and version of browser, operating system running browser

/

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220828120739.zip](https://github.com/OctoPrint/OctoPrint/files/9439526/octoprint-systeminfo-20220828120739.zip)

Mild case of the issue, hitting reload in the beginning of the video:
https://user-images.githubusercontent.com/5859228/187071222-fcccd693-0f28-4942-b72d-05e055870bb6.mov

foosel, 2023-01-25:
First of all sorry, this somehow got lost..

What I see up there which confuses me is this line just before it get stuck waiting for requests:

```
2022-08-28 12:01:34,144 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
```

It seems to be toggling back to online throughout the log, which will always trigger an attempt to refresh stuff, then realizes that it's offline after all and everything fails. That's obviously wrong, so I'll have to take a look at what make the connectivity checker decide it's online without having positive checks.

The way it should normally work is a) try a regular ping against a configured IP (by default I ship with 1.1.1.1) and b) try a regular name resolution for `octoprint.org`. A failure in *either* should lead to being marked as offline, and only when both succeed again should lead to being marked online.

cp2004, 2023-01-25:
Could part of the issues have been related to the missing timeouts fixed by https://github.com/OctoPrint/OctoPrint/issues/4539?

foosel, 2023-01-25:
No, that only affected the blacklist. And while I'm about to push a change to the timeouts of the requests we see timing out here (30s is WAY too high, switching to 3.05s (decimals because of the [TCP packet retransmission timer](https://www.ietf.org/rfc/rfc2988.txt)), it still doesn't explain the sudden switch back to online state, only to go back to offline right away. 

My current guess is that the DNS issues @crysxd had here were intermittent, so every here and there the connectivity check got back positive again, that triggered plugin manager, software update and announcement plugin (and whatever else was installed) to try to fetch whatever was needed, and in the case of the announcement plugin that doesn't request another connectivity check, whereas plugin manager and software update do. So the announcement plugin saw "we are online" and immediately try to fetch stuff from the net: each announcement channel, with a connect and read timeout of 30s. I'd expect a resolution error to not take the whole connect timeout to detect, but maybe there internal retries on the socket implementation or something that would explain that.

What I'm going to do here is to make sure the announcement channels look at the current online state before trying to work. That *still* leaves us a small window of stuff going down between check and fetch attempt, but with the timeout also changed to 3.05s) that should hopefully cause less issues. 

As far as I can see, what's going on here is a perfect storm created by *intermittent* issues that then cause requests to be attempted even though stuff is broken again right away, which then fail but with too large timeouts set so threads block too long and stuff grinds to a halt.

crysxd, 2023-01-25:
Thanks for checking this out @foosel! Not an important issue, so no worry about the delay.

The issue might very well have been intermittent as I'm 100% certain it was caused by Wifi issues. 

Just for my understanding, what's the reason for all those updates to run on the same thread as the  HTTP server? 

Because from my point of view, the actual problem is not that those update tasks were executed, nor that they took up to xx seconds to fail, but rather that while those update jobs were executed the web interface became unresponsive. If those updates would run on their own thread or would be triggered in a fire-and-forget manner (maybe with a check if the previous job is still running), this issue would not surface to the user.

Basically my problem was that I could not access the web interface due to the HTTP server being blocked while those updates were done.

foosel, 2023-01-25:
> Just for my understanding, what's the reason for all those updates to run on the same thread as the HTTP server?

They actually don't, Plugin Manager, Software Update and Announcements plugin all run these requests on their own threads:

- https://github.com/OctoPrint/OctoPrint/blob/68191cfbf439c14f96879c1fbc8e87ce46ae37cc/src/octoprint/plugins/pluginmanager/__init__.py#L626
- https://github.com/OctoPrint/OctoPrint/blob/68191cfbf439c14f96879c1fbc8e87ce46ae37cc/src/octoprint/plugins/softwareupdate/__init__.py#L1568-L1570
- https://github.com/OctoPrint/OctoPrint/blob/68191cfbf439c14f96879c1fbc8e87ce46ae37cc/src/octoprint/plugins/announcements/__init__.py#L308

But I could imagine here that a page load would block since that will try to fetch the software update/plugin manager/announcements data on load, and at least some of those would then block due to trying to wait for the result of the ongoing requests in the other threads. And since OctoPrint runs in Tornado, which is single threaded, that could block the whole server for other requests, until timeouts are triggered on those. I have not had a chance yet to try to simulate that scenario, but in any case it should only ever happen if the online check fails to detect the offline state or is disabled to begin with. And sadly there is no easy way to solve the blocking issue in this case without making a regular page load way more complicated than it should be.

And maybe that isn't even what happened and it was some other bundled plugin doing the blocking that I currently don't have on my radar 🤷‍♀️

crysxd, 2023-01-25:
Understood! Thanks for checking this issue out!

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel