OctoPrint/OctoPrint #4639: Mouseup makes dialogs close

napter, 2022-09-08:
### The problem

This is mostly an inconvenience. If you highlight text in a dialog and the mouse moves off the dialog before the mouseup the dialog closes. This is common since textboxes are often near the edges of dialogs. 

When it happens it is very confusing and despite knowing about the issue it still usually takes me a few tries before I realize what happened and then carefully select the text so that the final mouseup remains within the dialogs borders. 

The dialog should only close from a mousedown, not a mouseup.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

Windows

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

jneilliii, 2022-09-08:
I can confirm I've seen this before with some of my plugins that use bootstrap modal. I wonder if it's a potential issue with the library itself or if there's a setting that can be enabled on the modal call that would prevent this behavior.

omkarshinge, 2022-10-16:
- Regarding this issue I did a bit of research and found out that back then there are many bugs related to this issue in bootstrap source code . 
- By the way I'm new to the OctoPrint repository and found that version of bootstrap files used in the project are of v2.3.2 .
- Here is the link to one such bug filed in bootstrap framework's source code : [Link to the bug filed in v3 bootstrap source](https://github.com/twbs/bootstrap/issues/28844) .
- In above issue they have clearly stated that there won't be any fix provided by them for older versions (In the above issue it is mentioned v3.x) .
- There's workaround mentioned too by an user to change the modal dismissing part of code to listen to mousedown event instead of click .
- In newer version of Bootstrap (v5x)I  guess the issue is fixed : [Link to bug where issue is fixed](https://github.com/twbs/bootstrap/pull/36863) .
- So I think there are two options here either to upgrade bootstrap or to perform workaround and check if anything breaks due to the changes done .

cp2004, 2022-10-16:
> So I think there are two options here either to upgrade bootstrap or to perform workaround and check if anything breaks due to the changes done .

The first option (to upgrade bootstrap) is not an option for us - lots of discussion on this issue tracker & discord & probably the forums already as to why. We could make a change to fix the issue in the bundled bootstrap files to fix the issue or work around it.