OctoPrint/OctoPrint #4640: Flip webcam causes some Web UI to become not interactable.

pawelkw, 2022-09-10:
### The problem

Using the webcam feature with the following settings.
[image]
Causes this:
[image]
Elements obscured by this transparent HTML element are not clickable (so you can't access Temperature, Terminal, Settings, Power, etc). Deleting the associated tag and it's `<div>` parents manually in the browser dev mode resolves it. 

Tested browsers:
Safari: 15.5 (17613.2.7.1.8)
Firefox: 105.0b9 (64-bit

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.8.2 Python 3.8.13 

### Operating system running OctoPrint

Ubuntu Server

### Printer model & used firmware incl. version

n/a

### Browser and version of browser, operating system running browser

Firefox 105.0b9 (64-bit), MacOS 12.4

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220910093941.zip](https://github.com/OctoPrint/OctoPrint/files/9539967/octoprint-systeminfo-20220910093941.zip)

foosel, 2022-09-12:
@botflakes that doesn't sound like the same issue, and also like something you should rather report to the octolapse issue tracker.

botflakes, 2022-09-12:
@foosel Octolapse was just an example. Same thing happens with every other plugin. The only "popup" menu that loads normally is the Settingsmenu from OctoPrint and the Notificationcenter

foosel, 2022-09-12:
@botflakes Do you have the webcam flipped and rotated as stated by OP? If yes, then please test safe mode, if it persists provide a full system info bundle like OP. If you don't have the camera flipped/rotated this is not the same issue. If it vanishes in safe mode it's caused by a plugin.

botflakes, 2022-09-12:
Will test when the Print is done. I'll keep you updated

foosel, 2022-09-12:
@pawelkw After not being able to reproduce with MJPG I saw you have this happening with an HLS source. Can you please test if setting `https://test-streams.mux.dev/x36xhzz/x36xhzz.m3u8` as your webcam URL allows you to reproduce this? Because for me it doesn't.

pawelkw, 2022-09-15:
@foosel 
[image]

At first it worked correctly, right after setting it (including not blocking UI element). Then after a few reloads it doesn't stream anymore - black screen as you see here + the original bug is back (UI elements blocked as you see on the screenshot).

foosel, 2023-01-26:
I cannot reproduce this against 1.9.0.dev, no matter how often I reload. Webcam integration in 1.9.0 has been extracted into a plugin however, so maybe that automatically fixed it. I'll tag this as fixed for now, but with `needs testing`. Please, once 1.9.0rc1 hits, test this again on your end. If you can still reproduce it, we'd need to take a second look at this.

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel