OctoPrint/OctoPrint #4644: [Request] Configurable socket timeout or external way to serve (plugin) static files

moeffju, 2022-09-21:
### Is your feature request related to a problem? Please describe.

After installing a certain number of plugins with a number of static files, loading of these static files takes so long that the SockJs connection times out prematurely while the data and interface still load in the background, i.e. loading does not fail, it is just super slow.

[image] - already the staggering is visible, with ~4 files delivered each batch

I had installed a number of plugins, but the heaviest in terms of additional requests was the dashboard plugin. Somehow with the latest OctoPrint core update (1.8.3), the combination of plugins that worked okay before (on 1.8.2) has become slower to the point that I can not use OctoPrint outside of safe mode anymore. The plugins that made the most additional requests appeared to be the Dashboard plugin (lots of graphics) and for some reason the SpoolManager plugin (lots of events fired and js requests), but again, they worked ok on 1.8.2. Now loading consistently fails.

This is on a Raspberry Pi 3B using OctoPi, using the official RasPi Power Supply (but still occasionally getting undervolting, likely due to camera and fan activating at the same time - mostly no warning tho).

### Describe the solution you'd like

Ideally, requests to static files would be served outside the main Octoprint thread(s) that serve the websocket for control.

If that's not easily possible, a strict re-ordering of the requests might be possible, i.e. load the very core, establish the WS connection, then asynchronously load all the other files.

Alternatively, making the SockJs timeout configurable in the release version would be a workaround.

### Describe alternatives you've considered

I haven't looked into the core deep enough to know how the requests are handled and how much effort it would be to outsource those to a dedicated low-priority thread or something external like Caddy.

moeffju, 2022-09-21:
Relevant logs:

```
2022-09-21 10:43:09,232 - octoprint.plugins.octoeverywhere - INFO - Slipstream took 120.58294653892517 to fully update the cache
<<< [here I hit shift-reload in the browser] >>>
2022-09-21 10:45:05,338 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.180
2022-09-21 10:45:10,297 - octoprint.server.util.flask - INFO - Passively logging in user teal from ::ffff:192.168.0.180
2022-09-21 10:45:10,301 - octoprint.access.users - INFO - Logged in user: teal
2022-09-21 10:45:31,564 - octoprint.server.util.flask - INFO - Passively logging in user teal from ::ffff:192.168.0.180
2022-09-21 10:45:31,565 - octoprint.access.users - INFO - Logged in user: teal
<<< [around here the browser got the timeout message] >>>
2022-09-21 10:45:36,040 - octoprint.server.util.flask - INFO - Passively logging in user teal from ::ffff:192.168.0.180
2022-09-21 10:45:36,041 - octoprint.access.users - INFO - Logged in user: teal
2022-09-21 10:46:01,015 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.180
<<< [yet the sockjs still connects] >>>
```

cp2004, 2022-09-21:
You can actually configure the socket timeout already, under the development settings. I realised it wasn't listed in the documentation, so the link below is to the work-in-progress *new* documentation that is automatically generated from the settings schema. Updating the docs was regularly forgotten...

https://octoprint.github.io/docs/user-guide/configuration/config-yaml/#defaults_4

`devel.sockJSConnectTimeout`

You are seeing *very* long load times there - far higher than any I have seen or would expect. What is the load time in safe mode with no plugins installed? If it is longer than maybe 10 seconds (for me on a Pi 4 it is less than 2 seconds total), then there's something else slowing it down, and the plugins are making that problem much worse.

moeffju, 2022-09-21:
Yeah, I saw a reference to that setting in one of the old threads and tried it, but I think I didn't capitalize sockJS correctly. I'll try that again, thank you!

Indeed it is very slow and it has been faster before, if not incredibly fast. It looks like I cannot easily compare with safe mode now because many plugin (static) files appear to still be requested by the frontend (and served by the backend). Is there some sort of cache I can manually bust? Then I'm happy to log safemode first load and repeat load again.

I'm replacing the (old) official RasPi power supply with the new, boxy one in the hopes that the occasional undervolting alerts go away. I have checked the SD Card for bad blocks, found none. And I had low level formatted the card before flashing the OctoPi image on it, but I am thinking about maybe reformatting it once more. Otherwise, no clue what to check tbh. Pi is actively cooled, no heat throttling, SD card is a relatively speedy SanDisk one... I'm open to ideas :)

cp2004, 2022-09-21:
> Yeah, I saw a reference to that setting in one of the old threads and tried it, but I think I didn't capitalize sockJS correctly. I'll try that again, thank you!

To be fair, when I typed that out, I actually capitalized it wrong as well - going by the link I had it should be `sockJsConnectTimeout`.

If that doesn't make a difference, I was wondering if it was not the timeout you were hitting, and instead some other loading problem. That would not be explained easily by it working in safe mode, however.

> many plugin (static) files appear to still be requested by the frontend (and served by the backend)

When you are running in safe mode, you will still see `packed_plugins.js` and other associated files being requested because some of OctoPrint's core features are actually implemented as bundled plugins, which aren't disabled in safe mode. You shouldn't see the extra icon assets and the file sizes should be much much smaller. If you are still seeing those dashboard PNGs being requested, then I am confused.

moeffju, 2022-09-28:
I did still see them after restarting in safe mode, which also confused me. My 3D printer and OctoPi are currently partially disassembled for upgrades and stuff, so I cannot debug further right now, but anecdotally the issue has become better since I added the timeout - and funny enough it feels like loading times had also significantly improved. My best guess for now is that running the full badblocks scan on the SD card triggered internal rebalancing or something. (That, or the OctoEverywhere plugin was very incompatible with the new OctoPrint core -- there's been two updates for the plugin since I believe.)

In a few days I should have things running again, and then also with the new RasPi power supply (so no more undervolting, for good, I hope). Once it's back up, I'll measure again.