OctoPrint/OctoPrint #4645: Admin login (password) no longer works after updating to 1.8.3

bolsoncerrado, 2022-09-21:
### The problem

Just upgraded OctoPrint 1.8.2 Python 3.7.3 OctoPi 0.18.0 to Octoprint 1.8.3 and when restarting the server, I'm asked login/password.

Upon entering the all time working admin login and password, I'm told Incorrect username/password error.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Login failed

### Version of OctoPrint

OctoPrint 1.8.3 Python 3.7.3 

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Prusa MK3S

### Browser and version of browser, operating system running browser

Chrome 95 on MacOS BigSur

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

bolsoncerrado, 2022-09-21:
[image]

foosel, 2022-09-21:
Try the instructions behind the "Forgot password?" link. If anything there fails, please provide information as to how. Logs, screenshots, etc.

I've gone through several updates just yesterday during the final update tests, and countless before during development, none refused to login afterwards, so this is either a case of a forgotten username/password after all or somethings really odd going on. I'd also like to understand why I see a ton of successful logins in the log when it doesn't work according to you.

Also share the contents of your Javascript console when it goes wrong. 

> ### If you could not test in safe mode, please state why
> 
> Login failed

That is no reason to forgo testing in safe mode, that can be started perfectly fine via SSH, [see the docs](https://docs.octoprint.org/en/master/features/safemode.html). So please test safe mode too.

And since once again you initially didn't include the system info bundle or any other logs, those are not optional, never have been, never will be, and you've opened enough tickets by now that you should really know this.

fekle, 2022-09-21:
I'm having the same issue after updating to v1.8.3, albeit running the official octoprint docker image.
Everything was working fine, but after upgrading to v1.8.3 I can't log in anymore. The issue persists when starting in safe mode. 

The interesting thing is, even though my user is still present in `users.yaml`, but running `octoprint user list` just returns `0 users registered in the system:` - and running `octoprint user add` to add a new user succeeds, but doesn't change the `users.yaml` file and the new user is not able to log in.

Octoprint Version: 1.8.3
OS: Debian GNU/Linux 11 (bullseye)
Runtime: Docker 20.10.17, octoprint/octoprint:latest
Logs:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 01-mjpg-streamer-config: executing...
[cont-init.d] 01-mjpg-streamer-config: exited 0.
[cont-init.d] v3-dir-structure-check: executing...
AUTOMIGATE enabled....
[cont-init.d] v3-dir-structure-check: exited 0.
[cont-init.d] done.
[services.d] starting services
MJPG Streamer Version.: 2.0
[services.d] done.
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 960 x 720
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 o: www-folder-path......: /usr/local/share/mjpg-streamer/www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
2022-09-21 16:26:20,178 - octoprint.startup - INFO - ******************************************************************************
2022-09-21 16:26:20,182 - octoprint.startup - INFO - Starting OctoPrint 1.8.3
2022-09-21 16:26:20,183 - octoprint.startup - INFO - Starting in SAFE MODE. Third party plugins will be disabled!
2022-09-21 16:26:20,184 - octoprint.startup - INFO - Reason for safe mode: setting in config.yaml
2022-09-21 16:26:20,185 - octoprint.startup - INFO - ******************************************************************************
2022-09-21 16:26:20,863 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-09-21 16:26:20,877 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2022-09-21 16:26:20,878 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2022-09-21 16:26:20,878 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2022-09-21 16:26:21,085 - octoprint.startup - INFO - Blacklist processing done, adding 3 blacklisted plugin versions: roomtemp (any), GcodeEditor (>=0.1.1,<=0.2.8), gcodeleveling (>=0.1.0,<=0.1.1)
2022-09-21 16:26:21,105 - octoprint.plugin.core - INFO - Loading plugins from /octoprint/plugins/lib/python3.8/site-packages/octoprint/plugins, /octoprint/octoprint/plugins and installed plugin packages...
2022-09-21 16:26:24,081 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.6.13) did not pass check, not loading.
2022-09-21 16:26:24,302 - octoprint.plugin.core - INFO - Found 26 plugin(s) providing 17 mixin implementations, 34 hook handlers
2022-09-21 16:26:24,416 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2022-09-21 16:26:24,552 - py.warnings - WARNING - /octoprint/plugins/lib/python3.8/site-packages/flask_limiter/extension.py:317: UserWarning: Using the in-memory storage for tracking rate limits as no storage was explicitly specified. This is not recommended for production use. See: https://flask-limiter.readthedocs.io#configuring-a-storage-backend for documentation about configuring the storage backend.
  warnings.warn(
octoprint  |
2022-09-21 16:26:24,573 - octoprint.server - INFO - Intermediary server started
2022-09-21 16:26:24,603 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /octoprint/octoprint/uploads...
2022-09-21 16:26:24,997 - octoprint.filemanager.storage - INFO - ... file metadata for /octoprint/octoprint/uploads initialized successfully.
2022-09-21 16:26:25,007 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2022-09-21 16:26:25,007 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2022-09-21 16:26:25,008 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2022-09-21 16:26:25,010 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2022-09-21 16:26:25,011 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')")
[... cut ...]
2022-09-21 16:26:39,125 - octoprint.plugins.softwareupdate - INFO - Minimum free storage across all update relevant locations is 49.8GB. That is considered sufficient for updating.
2022-09-21 16:26:39,715 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'http://REDACTED/', 'path': '/', 'query_string': 'l10n=en'}
2022-09-21 16:26:40,210 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2022-09-21 16:26:53,214 - octoprint.server.preemptive_cache - INFO - ... done in 13.50s
```

EzeCuervo, 2022-09-21:
Hi all! 
I got the same error as bolsoncerrado after upgraded from OctoPrint 1.8.2 to Octoprint 1.8.3, I can't login with my credentials. I checked the console log and I'm receiving the following error:

`
POST https://xxx.xxx/api/login 400
jquery.min.js:2 
send	@	jquery.min.js:2
ajax	@	jquery.min.js:2
OctoPrintClient.ajax	@	packed_client.js?8cc16385:12
OctoPrintClient.ajaxWithData	@	packed_client.js?8cc16385:12
OctoPrintClient.post	@	packed_client.js?8cc16385:12
OctoPrintClient.postJson	@	packed_client.js?8cc16385:12
OctoPrintBrowserClient.login	@	packed_client.js?8cc16385:59
(anonymous)	@	login.js:36
dispatch	@	jquery.min.js:2
v.handle	@	jquery.min.js:2
`

I have my OctoPrint behind a proxy (Nginx Proxy Manager) so I tried login in avoiding the proxy connection (using local ip address) and it works fine.

My Nginx Proxy Manager configuration for the OctoPrint domain is:
[image]

Regards!

foosel, 2022-09-21:
@EzeCuervo @fekle OctoPi also ships with a reverse proxy, haproxy in fact, and there it works just fine. Are you using a custom script root ("example.com/octoprint/") and forgot to set the corresponding [reverse proxy header `X-Script-Name`](https://community.octoprint.org/t/reverse-proxy-configuration-examples/1107) by any chance? If so, OctoPrint's frontend won't be able to find the correct cookie value to inject, failing the CSRF check.

fekle, 2022-09-21:
Hi, thanks for your reply @foosel.
No im just using an standard haproxy setup as a transparent reverse proxy with a dedicated subdomain for octoprint and no sub path. It works fine for all other things i host and was working before the octoprint update. But it could very well be that i habe to change something in my configuration, maybe its the basic auth that im using - will check when i get back home and let you know.

foosel, 2022-09-21:
Could it be that in the non working setups the reverse proxy sits in front of OctoPi's own reverse proxy (that's what the nginx proxy manager screenshot suggests), so we are actually looking at a double reverse proxy scenario?

Then again, I just tested this precise scenario and that works flawlessly here. Nginx, even with custom script root, in front of OctoPi's haproxy, in front of OctoPrint 1.8.3. Login works, no CSRF mismatches.

EzeCuervo, 2022-09-21:
> @EzeCuervo @fekle OctoPi also ships with a reverse proxy, haproxy in fact, and there it works just fine. Are you using a custom script root ("example.com/octoprint/") and forgot to set the [...]

Hi @foosel! Thanks for your quick reply.
I'm just using Nginx Proxy Manager with a dedicated subdomain and no sub path as @fekle. Before the update it was working fine but in my case i'm not using basic auth with the Nginx Proxy Manager. I'll keep debugging if I have the chance to solve the issue. The weird thing is that before the update was working like a charm. :(

bolsoncerrado, 2022-09-21:
In my case it fails to log in when (still) using frameset on Opera 70.0 (which I stopped updating due to the previous login issues I was having).

If I log in a separate window (dedicated) it works, but something else must be failing because once logged in, if I return to the frameset environment it fails again.

foosel, 2022-09-21:
@EzeCuervo @fekle 

I just configured a custom (internal) domain to have a subdomain pointing at one of my test instances behind OctoPi's haproxy. So, `https://octopia.fsl.cx` -> `http://octopia.lan` -> `http://127.0.0.1:5000`. No issue. Whatever this is, it is something very specific to y'all's setup.

@fekle could you maybe share the (redacted) config you are using? I know it's hard to get this out of this weird Nginx Proxy Manager thing, hence I'm asking you and not @EzeCuervo, though if I could get an actual nginx config shared by you as well, that would be lovely. I can then see if I can maybe reproduce it THAT way. I'm pretty sure your problem is in fact being caused by some issue of the client to grab the cookie with the CSRF token it's supposed to put into a custom header (you can check the login POST request, it should contain both a `csrf_token_P<something>` cookie *and* a `X-CSRF-Token` header), but I currently fail to see as to why and cannot trigger this myself in any of my various setups here.

@bolsoncerrado 

> If I log in a separate window (dedicated) it works, but something else must be failing because once logged in, if I return to the frameset environment it fails again.

Ah... You might remember that I said that framesets are no longer supported and stuff would soon break due to `SameSite` rules a while back, and warned you that stuff could break anytime? Alongside the CSRF protection I also rolled out the way more secure `SameSite Strict` ruling now, as `Lax` would likely break soon in browsers anyhow and is outdated. You could try setting `server.cookies.samesite` to `lax` in your `config.yaml`, but rest assured that your frameset setup *will* at some point irreparably break, it simply does not match modern security settings in browsers and will soon be forced out by firefox, chrome etc.

Relevant section in the docs: https://docs.octoprint.org/en/master/configuration/config_yaml.html#server

EzeCuervo, 2022-09-21:
Sure @foosel, let me share more information with you:

I'm using Cloudflare for my domain with DNS proxy and the SSL/TLS encryption mode is set to Flexible. 

`https://subdomain.domain.com`  -> `Nginx Proxy Manager`  -> `http://10.10.1.10`  (dummy IP address)

Here is the Nginx Proxy Manager config file for this subdomain:

A

```
server {
  set $forward_scheme http;
  set $server         "10.10.1.10";
  set $port           80;

  listen 80;
listen [::]:80;

  server_name subdomain.domain.com;

# Asset Caching
  include conf.d/include/assets.conf;

  # Block Exploits
  include conf.d/include/block-exploits.conf;

proxy_set_header Upgrade $http_upgrade;
proxy_set_header Connection $http_connection;
proxy_http_version 1.1;

  access_log /data/logs/proxy-host-10_access.log proxy;
  error_log /data/logs/proxy-host-10_error.log warn;

  location / {
 
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $http_connection;
    proxy_http_version 1.1;
    
    # Proxy!
    include conf.d/include/proxy.conf;
  }

  # Custom
  include /data/nginx/custom/server_proxy[.]conf;
}
```

foosel, 2022-09-21:
New guess: You are accessing OctoPrint through https via Cloudflare, but OctoPrint doesn't know that and thus generates the cookie name for port 80 (people are running several instances on one host, so I had to force port and path specific cookie names to not cause mayhem in such shared setups).

Your nginx config certainly is missing these headers:

```
proxy_set_header X-Real-IP $remote_addr;
proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
proxy_set_header X-Scheme $scheme;
```

which could lead to misunderstandings between OctoPrint's front- and backend as to what the cookie names should be. If you *only* access OctoPrint through that reverse proxy chain at all times, you could also go about this with config settings, see `server.reverseProxy.{prefix|scheme|host}Fallback` in https://docs.octoprint.org/en/master/configuration/config_yaml.html#server

cp2004, 2022-09-21:
Using the reverse proxy configuration examples can also be useful to compare.

https://community.octoprint.org/t/reverse-proxy-configuration-examples/1107

fekle, 2022-09-21:
So I was able to investigate this a bit further - the issue is indeed in the CSRF validation.

I was able to fix the issue by adding `http-request add-header X-Scheme "https"` to my HAProxy configuration. Strange that it worked without this before, but well there you have it :) 

Thanks for your help!

For the sake of completeness, some more information/answers:

- I'm not using CloudFlare or anything else in front of my server
- I am using my own HAProxy in addition to OctoPrint's own as i am using this server/IP-Address for more than just OctoPrint.
- I tried bypassing OctoPrint's own HAProxy by proxying directly to the python server at port 5000, but the same problem occurs (Invalid CSRF Token)
- I am using the `option forwardfor` option in HAProxy to forward the client ip address.
- OctoPrint is not logging any errors to its log file
- The server response upon a login request is `400 Bad Request` with a body of `{"error":"CSRF validation failed"}`.

The redacted haproxy config:
```haproxy
global
    uid                         80
    gid                         80
    chroot                      /var/haproxy
    daemon
    stats                       socket [REDACTED] group proxy mode 775 level admin expose-fd listeners
    nbproc                      1
    nbthread                    4
    hard-stop-after             60s
    no strict-limits
    tune.ssl.default-dh-param   2048
    spread-checks               2
    tune.bufsize                16384
    tune.lua.maxmem             0
    log                         /var/run/log local0 info
    lua-prepend-path            /tmp/haproxy/lua/?.lua

defaults
    log     global
    option redispatch -1
    timeout client 30s
    timeout connect 30s
    timeout server 30s
    retries 3
    default-server init-addr last,libc

frontend default
    bind 0.0.0.0:8443 name 0.0.0.0:8443 ssl alpn h2,http/1.1 crt-list [REDACTED]
    mode http
    option http-keep-alive
    option forwardfor
    timeout client 30s

    acl [REDACTED] hdr(host) -i octoprint.[REDACTED]

    use_backend octoprint if [REDACTED]

    http-response set-header Strict-Transport-Security "max-age=31556952; includeSubDomains; preload;"

backend octoprint
    mode http
    balance roundrobin
    stick-table type ip size 50k expire 30m

    timeout connect 30s
    timeout server 30s
    http-reuse safe
    server octoprint [REDACTED]:80 check inter 2s

    http-request add-header X-Scheme "https"  # <-- IMPORTANT!
```

EzeCuervo, 2022-09-22:
Thanks @cp2004 for your reply, too!

> New guess: You are accessing OctoPrint through https via Cloudflare, but OctoPrint doesn't know that and thus generates the cookie name for port 80 (people are running several instances on one host, [...]

I'm still having the issue... I added these lines in my Nginx Proxy Manager config, leaving the location / as:
```
  location / {
    proxy_set_header Host $host;
    proxy_set_header X-Forwarded-Scheme $scheme;
    proxy_set_header X-Forwarded-Proto  $scheme;
    proxy_set_header X-Forwarded-For    $remote_addr;
    proxy_set_header X-Real-IP		$remote_addr;
    proxy_pass       http://10.10.1.10:80;
    
    # Block Exploits
    include conf.d/include/block-exploits.conf;

    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $http_connection;
    proxy_http_version 1.1;
    
    proxy_set_header X-Real-IP $remote_addr;
    proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    proxy_set_header X-Scheme $scheme;
  }
```

Then I read the config.yaml documentation, I added the following lines but i'm a little bit confused on how I have to declare these things:

```
  reverseProxy:
    hostHeader: X-Forwarded-Host
    prefixHeader: X-Script-Name
    schemeHeader: X-Scheme
```

Also, I have enabled the CORS option. In case you want to close this issue, let me know which is the best channel to continue troubleshooting.

Thanks again!
Eze

foosel, 2022-09-22:
@EzeCuervo 

> Then I read the config.yaml documentation, I added the following lines but i'm a little bit confused on how I have to declare these things:

You don't have to declare those things. Those are the default values used by OctoPrint. BUT you can prefill the fallback values used in case of absence of the headers in the incoming requests via the `*Fallback` config settings, e.g.

```yaml
server:
  reverseProxy:
    hostFallback: example.com
    schemeFallback: https
```

I'm talking about these:

[image]

Looking at your nginx config, there's some stuff doubled now and I don't know how that will affect things. So let's clean this up a bit:

```
  location / {
    proxy_set_header Host               $http_host;
    proxy_set_header X-Forwarded-Scheme $scheme;
    proxy_set_header X-Forwarded-Proto  $scheme;
    proxy_set_header X-Scheme           $scheme;
    proxy_set_header X-Forwarded-For    $proxy_add_x_forwarded_for;
    proxy_set_header X-Real-IP		$remote_addr;
    proxy_pass       http://10.10.1.10:80;
    
    # Block Exploits
    include conf.d/include/block-exploits.conf;

    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $http_connection;
    proxy_http_version 1.1;
  }
```

Note that I switched `Host` to `$http_host`, so it will *only* be taken from the incoming `Host` header. It could be that is also a factor here, I can't remember why I originally used `$http_host` instead of `$host` sadly.

I don't know how Cloudflare's SSL termination works here, and I'm currently not sure if Nginx's `$scheme` is filled based on the incoming request or on the port the request was received on. So since your Nginx is listening only on port 80 and thus http, you might also want to set the header value explicitly to https in Nginx and see if that fixes things:

```
  location / {
    proxy_set_header Host               $http_host;
    proxy_set_header X-Forwarded-Scheme https;
    proxy_set_header X-Forwarded-Proto  https;
    proxy_set_header X-Scheme           https;
    proxy_set_header X-Forwarded-For    $proxy_add_x_forwarded_for;
    proxy_set_header X-Real-IP		$remote_addr;
    proxy_pass       http://10.10.1.10:80;
    
    # Block Exploits
    include conf.d/include/block-exploits.conf;

    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $http_connection;
    proxy_http_version 1.1;
  }
```

> Also, I have enabled the CORS option.

That's going to do nothing here.

@fekle 

> Strange that it worked without this before, but well there you have it :)

Pure luck ;) This could have broken by any other change in OctoPrint related to header evaluation, cookie management etc as well, it now broke on the new CSRF protection but it was always misconfigured. Glad you got it sorted!

EzeCuervo, 2022-09-23:
@foosel 
> You don't have to declare those things. Those are the default values used by OctoPrint. BUT you can prefill the fallback values used in case of absence of the headers in the incoming requests via the `*Fallback` config settings, e.g.

LOL, I thought that I don't have to declare those things but when you are going crazy with an issue, you tried things that have no sense. I configured th

> 
> ```yaml
> server:
>   reverseProxy:
>     hostFallback: example.com
>     schemeFallback: https
> ```

I updated the config file with this but nothing changes.

> Looking at your nginx config, there's some stuff doubled now and I don't know how that will affect things. So let's clean this up a bit:
>
> ```
>   location / {
>     proxy_set_header Host [...]

Yes, the duplicated parameters are set by Nginx Proxy Manager. I tried to edit the config file manually but when I disable and enable the domain again, NPM adds the first parameters:
 ```
    proxy_set_header Host $host;
    proxy_set_header X-Forwarded-Scheme $scheme;
    proxy_set_header X-Forwarded-Proto  $scheme;
    proxy_set_header X-Forwarded-For    $remote_addr;
    proxy_set_header X-Real-IP		$remote_addr;
 ```

> I don't know how Cloudflare's SSL termination works here, and I'm currently not sure if Nginx's `$scheme` is filled based on the incoming request or on the port the request was received on. So since [...]

Same here, I gave up trying to make NPM work again with OctoPrint, I'll look up another idea to publish my OctoPrint instance.

Once again, thanks for all your help Gina. You rock!
Eze

foosel, 2022-09-27:
Since this looks like it was caused by either `SameSite: Strict` or reverse proxy misconfigurations, and 1.8.4 that was just released sets `SameSite: Lax` and also includes a new revers proxy test page to debug misconfigurations, I'm closing this. I don't see anything else actionable here in any case.

closed 2022-09-27 by foosel