OctoPrint/OctoPrint #4647: OctoPrint Docs Don't List All Possible Values From `printer.get_state_id()`

QuinnDamerell, 2022-09-21:
### The problem

After doing some debugging, it looks like `get_state_id()` on the `printer` object returns more values than are listed in the docs. The biggest problem I hit was that it seems to return `resuming`, and I had logic that was checking if the system was in a printing state but only looking for `printing`. (where `resuming` and `finishing` I would consider "printing state").

I would make a PR and update the docs, but I couldn't follow the logic confidently enough to assert what all of the possible states are.

P.S. The docs also seem to be missing a few other state functions, like `is_resuming()`, I'm not sure if that's intentional. 

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

Latest

### Operating system running OctoPrint

N/A

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2022-09-22:
Definitely not intentional and simply an oversight on my part. Didn't sync up stuff.

Ajimaru, 2026-01-07:
This issue has been resolved in the `dev` branch as of July 1, 2025 ([commit 5669e8f9d](https://github.com/OctoPrint/OctoPrint/commit/5669e8f9d)).

**What was changed:**

The documentation for `get_state_id()` now correctly references the `ConnectedPrinterState` enum class instead of manually listing state values. This ensures the documentation automatically stays in sync with the actual implementation.

**Current state values** (as of dev branch):

The `ConnectedPrinterState` enum in [src/octoprint/printer/connection.py](https://github.com/OctoPrint/OctoPrint/blob/dev/src/octoprint/printer/connection.py#L40-L56) defines 14 states:
- `DETECTING`
- `CONNECTING`
- `OPERATIONAL`
- `STARTING`
- `PRINTING`
- `PAUSING`
- `PAUSED`
- `RESUMING`
- `CANCELLING`
- `FINISHING`
- `CLOSED`
- `ERROR`
- `CLOSED_WITH_ERROR`
- `TRANSFERRING_FILE`

**Why the online docs still show the old version:**

The fix is currently only in the `dev` branch and hasn't been included in a release yet (the latest release is 1.11.5 from November 2025). The updated documentation will appear on docs.octoprint.org once the next version is released.

The issue can be closed as resolved, with the fix pending publication in the next release.