OctoPrint/OctoPrint #4648: Passive login with API key fails in 1.8.3

crysxd, 2022-09-22:
### The problem

After the update to 1.8.3 the `/api/login` request returns 500. The below sample request works fine in 1.8.2 and is used by OctoApp to login for a web socket connection:

```
--> POST http://192.168.1.31:5004/api/login http/1.1
X-Api-Key: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Accept: application/json
Content-Type: application/json; charset=UTF-8
Content-Length: 16
Host: 192.168.1.31:5004
Connection: Keep-Alive
Accept-Encoding: gzip
User-Agent: okhttp/4.9.1
{"passive":true}
--> END POST (16-byte body)
<-- 500 INTERNAL SERVER ERROR http://192.168.1.31:5004/api/login (107ms)
Content-Type: application/json
Content-Length: 167
Cache-Control: no-store, no-cache, must-revalidate, post-check=0, pre-check=0, max-age=0
Pragma: no-cache
Expires: -1
X-Clacks-Overhead: GNU Terry Pratchett
Server-Timing: app;dur=83
X-Robots-Tag: noindex, nofollow, noimageindex
X-Content-Type-Options: nosniff
X-Frame-Options: sameorigin
{"error":"The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."}
<-- END HTTP (167-byte body)
```

After encountering issues on some of my test instances I performed following steps:

- On an instance with 1.8.2 confirm the above request works fine and returns a session
- Update to 1.8.3 using the webinterface
- Repeat the above request, now results in 500

The logs show following error for each time the request is made:

```
2022-09-22 04:54:55,665 - octoprint.server.util.sockjs - INFO - Client connection closed: 172.18.0.1
2022-09-22 04:54:56,872 - octoprint.server.util.sockjs - INFO - New connection from client: 172.18.0.1
2022-09-22 04:54:56,905 - octoprint.server.util.flask - INFO - Passively logging in user _api from 172.18.0.1
2022-09-22 04:54:56,905 - octoprint.access.users - INFO - Logged in user: _api
2022-09-22 04:54:56,926 - octoprint - ERROR - Exception on /api/login [POST]
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/octoprint/plugins/lib/python3.8/site-packages/flask/app.py", line 1519, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/octoprint/plugins/lib/python3.8/site-packages/flask/app.py", line 1517, in full_dispatch_request
    rv = self.dispatch_request()
  File "/octoprint/plugins/lib/python3.8/site-packages/flask/app.py", line 1503, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/octoprint/plugins/lib/python3.8/site-packages/flask_limiter/extension.py", line 1145, in __inner
    R, flask.current_app.ensure_sync(cast(Callable[P, R], obj))(*a, **k)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/api/__init__.py", line 357, in login
    return passive_login()
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 730, in passive_login
    user = login(determine_user(user))
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 676, in login
    flask.session["usersession.signature"] = session_signature(
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1911, in session_signature
    key = userManager.signature_key_for_user(user, salt=current_app.config["SECRET_KEY"])
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/access/users.py", line 814, in signature_key_for_user
    raise UnknownUser(username)
 octoprint.access.users.UnknownUser: Unknown user: _api
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.3

### Operating system running OctoPrint

Docker

### Printer model & used firmware incl. version

/

### Browser and version of browser, operating system running browser

/

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220922045824.zip](https://github.com/OctoPrint/OctoPrint/files/9622401/octoprint-systeminfo-20220922045824.zip)

foosel, 2022-09-22:
Good news, I know what's happening and that should hopefully be an easy fix. I'm now debating however whether I should indeed fix it or just get rid of the global API key altogether at this opportunity.

crysxd, 2022-09-22:
I vote for the second option. But then this endpoint should still return something else than 500, ideally 403 so clients will handle it as if the API key would be invalid (which it indeed is)

crysxd, 2022-09-22:
@foosel I think there is another issue with this endpoint. 

1.8.2 responds 200 OK with an invalid API key and a guest user:
```
{
    "_is_external_client": false,
    "needs": {
        "group": [
            "guests"
        ]
    }
}
```

1.8.3 responds with 400:
```
{
    "error": "CSRF validation failed"
}
```

I'm not sure if this is intended, but I'd also expect a 403 in case the API key is invalid

foosel, 2022-09-22:
That changes things to "this requires a fix". 1.8.4 it is then.

foosel, 2022-09-22:
So, the fix for the UnknownUser error for the global API user is fixed by the above commit and ready for 1.8.4. The behaviour with an invalid API key still needs to be checked.

crysxd, 2022-09-22:
Will check the API key stuff, give me 15 minutes

foosel, 2022-09-22:
`staging/bugfix` branch ^^

crysxd, 2022-09-22:
It still seems to behave the same way regarding the wrong API key, but the global one is fine (pointed software update to the `staging/bugfix`).

Basically this is my test case:

```
curl --location --request POST 'http://octopi.local/api/login' \
--header 'X-Api-Key: iamwrong' \
--header 'Content-Type: application/json' \
--data-raw '{
    "passive": "true"
}'
```

(not trying to steal you check authority :D)

crysxd, 2022-09-22:
@foosel This seems now to work as expected but I now also `/api/currentuser` returns 403 instead of the guest user when provided with an invalid API key. I guess this is intended?

foosel, 2022-09-22:
Yeah, invalid API key should always lead to a 403. That it didn't before was a bug. Missing API key should be treated as guest.

crysxd, 2022-09-22:
Perfect :) Then I'll create an update as well to account for that.

Thanks a lot for the quick fixes!!

spanzetta, 2022-09-24:
I am not sure if this issue is also the cause of the problem accessing to Octoprint from Octoapp (https://play.google.com/store/apps/details?id=de.crysxd.octoapp&hl=it&gl=US) when accessing from external network (with Router port redirection, which was always working until previous Octoprint build) but this is what I am experiencing after the update to 1.8.3.
I get the error message "Octoapp can communicate with xxxx.xxxxx.net but the websocket failed to connect with response code 500"
Will this also be fixed after the next update?
Thanks

crysxd, 2022-09-24:
That's the exact issue. Reconnect OctoPrint and when prompted use the "Give Access" button in the webinterface instead of the global API key.

spanzetta, 2022-09-24:
I already did that way and it works.. but only when using Internal IP/Wifi connection..
When using "external access" (using dyndns name and router port redirection.. which was always working with previous Octoprint builds) I get the error about "response code 500"..
Can it be another issue?

crysxd, 2022-09-24:
Hard to say without details...can you use main menu > settings > FAQ & help > Report a bug after you saw the 500? Then we can check the app logs and see what's going on. (But then let's not spam this issue here as it might not be related)

jneilliii, 2022-09-24:
> When using "external access" (using dyndns name and router port redirection.. which was always working with previous Octoprint builds) I get the error about "response code 500".. Can it be another issue?

There is another issue #4645 for stuff related to port forwarding and reverse proxy config that you may need to tweak to correct your issue @spanzetta.

foosel, 2022-09-24:
Not quite on topic here, but it must be said: Stop. With. Port Forwards. Port Forwards are a security concern and at this point considered a misconfiguration. A 3d printer does not belong on the public internet.

commit b87b965f (2022-09-22): 🐛 Fix global API key vs session sigs

Closes #4648
closed 2022-09-27 by foosel