OctoPrint/OctoPrint #4649: Print time estimation totally wrong in 1.8.3

otto2704, 2022-09-22:
### The problem

Just upgraded to 1.8.3 (from 1.8.2). Now the print estimation shows 56 hours remaining for a 5h print. Before it worked.
Using SuperSlicer 2.5.59

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

printing now

### Version of OctoPrint

1.8.3

### Operating system running OctoPrint

OctoPi 0.18

### Printer model & used firmware incl. version

Prusa MK3S+

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220922173918.zip](https://github.com/OctoPrint/OctoPrint/files/9627376/octoprint-systeminfo-20220922173918.zip)

foosel, 2022-09-22:
Print time estimation wasn't even remotely touched between 1.8.2 and 1.8.3, please read the changelog. 

Also you are using the Print Time Genius plugin, so if the offending print has been analysed with that, it's the plugin that's reporting wrong data here.

> ### If you could not test in safe mode, please state why
> 
> printing now

No, that is not excuse to not test in safe mode, that is only an excuse for waiting with a bug report until you *can* test in safe mode.

otto2704, 2022-09-22:
Yes, I am using Print Time Genius. Strange that the estimation worked without problems in 1.8.2.
Do you REALLY think, safe mode would change anything?

foosel, 2022-09-22:
Considering that Print Time Genius would be disabled, at the very least this might give you an additional data point. 

Check if the print has been adjusted by PTG (yellow star next to the name AFAIK). Check safe mode. If things still don't work feel free to ask for support on the community forums or on Discord. But given that as I said nothing on the native print time estimation was touched between 1.8.2 and 1.8.3, this is in all likelihood not a bug in stock OctoPrint. Happy to be proven otherwise over the course of a support session on the forums or Discord and then to reopen this here, but until then I'm closing this as - again - nothing in there was changed between 1.8.2 and 1.8.3, stuff is way to finicky even when it works just fine due to the nature of estimating a process like this, and you have a third party plugin in the mix that definitely DOES mess with the very process you are having issues with here.

Please don't take this the wrong way, but I've sunk WEEKS into debugging print time estimation bugs that then turned out to be issues on the user's side or in their workflow, so this is now something where I expect a proof first that it is indeed core OctoPrint.

closed 2022-09-22 by foosel

otto2704, 2022-09-22:
Hi!
Thanks for your answer. will try as soon as my print finishes. this can take some time ;-)
Thanks for your work!