OctoPrint/OctoPrint #4650: Remember Me fails with http after closing and re-opening browser a couple of times OctoPrint 1.8.3

jneilliii, 2022-09-23:
### The problem

When using the remember me option in OctoPrint 1.8.3 on no-secure http the login fails after closing browser a couple of times. Steps to reproduce:

1. open browser go to http://<ip>/
2. login with remember me checked
3. close browser
4. open browser go to http://<ip>/ (successful login)
5. close browser
6. open browser go to http://<ip>/ (failure with error)

[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.3

### Operating system running OctoPrint

octopi 0.18

### Printer model & used firmware incl. version

virtual printer

### Browser and version of browser, operating system running browser

Chrome Version 105.0.5195.127 (Official Build) (64-bit), windows

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220922212731.zip](https://github.com/OctoPrint/OctoPrint/files/9630246/octoprint-systeminfo-20220922212731.zip)

Additional details and trouble-shooting steps posted on Discord: https://discord.com/channels/704958479194128507/1022573961235664917

jneilliii, 2022-09-23:
Issue seems to only be impacting http (80), https (443) is working as expected.

foosel, 2022-09-26:
What I can see is that on the second open there's a cache hit, and stepping through the response generation at this point doesn't add the CSRF cookie header to the response, probably because the response isn't an OctoPrintFlaskResponse. 

Not clear why that only affects http and not https, but that's definitely going to cause issues like this.

foosel, 2022-09-26:
I think I've found it. At least I no longer can reproduce it now. Can anyone confirm?

jneilliii, 2022-09-26:
I'll check against that commit tonight.

jneilliii, 2022-09-26:
Actually forgot how easy it was to switch to commit release. Changed to bugfix/staging branch and updated and am no longer able to reproduce the issue.

commit daa5dbc4 (2022-09-26): 🐛 Fix setting CSRF cookie on cached responses

In case of a "Not modified" response, the CSRF cookie
wasn't added, leading to a loading failure.

Closes #4650
closed 2022-09-27 by foosel