OctoPrint/OctoPrint #4652: OctoPrint failed to update from 1.8.2 to 1.8.3 using builtin update plugin

derMart, 2022-09-25:
### The problem

Trying to update octoprint using the builtin updater plugin (so no third party plugin should be involved here).
Clicking on the toast to update to 1.8.3 gives a lot of errors (see attached log file).
I suppose the culprit is the installation / building of cffi:
```
2022-09-25 13:38:26,247 > Downloading argon2-cffi-bindings-21.2.0.tar.gz (1.8 MB)
2022-09-25 13:38:30,458 > Installing build dependencies: started
2022-09-25 13:39:32,379 > Installing build dependencies: still running...
2022-09-25 13:39:39,718 > Installing build dependencies: finished with status 'error'
2022-09-25 13:39:39,720 > Downloading argon2-cffi-bindings-21.1.0.tar.gz (1.8 MB)
2022-09-25 13:39:40,250 ! ERROR: Command errored out with exit status 1:
2022-09-25 13:39:40,251 ! command: /home/martin/OctoPrint/venv/bin/python3 /tmp/pip-standalone-pip-c42zk1ui/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-d_c6qy4r/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=45' 'setuptools_scm>=6.2' wheel 'cffi>=1.0.1'
2022-09-25 13:39:40,252 ! cwd: None
2022-09-25 13:39:40,253 ! Complete output (114 lines):
2022-09-25 13:39:40,255 ! Collecting setuptools>=45
2022-09-25 13:39:40,256 ! Downloading setuptools-65.4.0-py3-none-any.whl (1.2 MB)
2022-09-25 13:39:40,257 ! Collecting setuptools_scm>=6.2
2022-09-25 13:39:40,258 ! Downloading setuptools_scm-7.0.5-py3-none-any.whl (42 kB)
2022-09-25 13:39:40,260 ! Collecting wheel
2022-09-25 13:39:40,261 ! Downloading wheel-0.37.1-py2.py3-none-any.whl (35 kB)
2022-09-25 13:39:40,262 ! Collecting cffi>=1.0.1
2022-09-25 13:39:40,263 ! Downloading cffi-1.15.1.tar.gz (508 kB)
2022-09-25 13:39:40,264 ! Preparing metadata (setup.py): started
2022-09-25 13:39:40,265 ! Preparing metadata (setup.py): finished with status 'done'
2022-09-25 13:39:40,266 ! Collecting typing-extensions
2022-09-25 13:39:40,267 ! Downloading typing_extensions-4.3.0-py3-none-any.whl (25 kB)
2022-09-25 13:39:40,268 ! Collecting tomli>=1.0.0
2022-09-25 13:39:40,270 ! Downloading tomli-2.0.1-py3-none-any.whl (12 kB)
2022-09-25 13:39:40,271 ! Collecting packaging>=20.0
2022-09-25 13:39:40,272 ! Downloading packaging-21.3-py3-none-any.whl (40 kB)
2022-09-25 13:39:40,273 ! Collecting pycparser
2022-09-25 13:39:40,274 ! Downloading pycparser-2.21-py2.py3-none-any.whl (118 kB)
2022-09-25 13:39:40,275 ! Collecting pyparsing!=3.0.5,>=2.0.2
2022-09-25 13:39:40,276 ! Downloading pyparsing-3.0.9-py3-none-any.whl (98 kB)
2022-09-25 13:39:40,278 ! Building wheels for collected packages: cffi
2022-09-25 13:39:40,279 ! Building wheel for cffi (setup.py): started
2022-09-25 13:39:40,280 ! Building wheel for cffi (setup.py): finished with status 'error'
2022-09-25 13:39:40,281 ! ERROR: Command errored out with exit status 1:
2022-09-25 13:39:40,282 ! command: /home/martin/OctoPrint/venv/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-4b8f_q2p/cffi_76869bb599674e5aa5db38681ba78ded/setup.py'"'"'; __file__='"'"'/tmp/pip-install-4b8f_q2p/cffi_76869bb599674e5aa5db38681ba78ded/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-gm1rh4zo
2022-09-25 13:39:40,283 ! cwd: /tmp/pip-install-4b8f_q2p/cffi_76869bb599674e5aa5db38681ba78ded/
2022-09-25 13:39:40,284 ! Complete output (36 lines):
2022-09-25 13:39:40,286 ! running bdist_wheel
2022-09-25 13:39:40,287 ! running build
2022-09-25 13:39:40,288 ! running build_py
2022-09-25 13:39:40,289 ! creating build
2022-09-25 13:39:40,290 ! creating build/lib.linux-armv7l-3.8
2022-09-25 13:39:40,291 ! creating build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,292 ! copying cffi/verifier.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,293 ! copying cffi/vengine_gen.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,294 ! copying cffi/vengine_cpy.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,296 ! copying cffi/setuptools_ext.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,297 ! copying cffi/recompiler.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,298 ! copying cffi/pkgconfig.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,300 ! copying cffi/model.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,301 ! copying cffi/lock.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,302 ! copying cffi/ffiplatform.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,303 ! copying cffi/error.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,304 ! copying cffi/cparser.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,306 ! copying cffi/commontypes.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,307 ! copying cffi/cffi_opcode.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,308 ! copying cffi/backend_ctypes.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,309 ! copying cffi/api.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,310 ! copying cffi/__init__.py -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,311 ! copying cffi/_cffi_include.h -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,312 ! copying cffi/parse_c_type.h -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,314 ! copying cffi/_embedding.h -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,315 ! copying cffi/_cffi_errors.h -> build/lib.linux-armv7l-3.8/cffi
2022-09-25 13:39:40,316 ! running build_ext
2022-09-25 13:39:40,317 ! building '_cffi_backend' extension
2022-09-25 13:39:40,318 ! creating build/temp.linux-armv7l-3.8
2022-09-25 13:39:40,319 ! creating build/temp.linux-armv7l-3.8/c
2022-09-25 13:39:40,320 ! arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DFFI_BUILDING=1 -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/home/martin/OctoPrint/venv/include -I/usr/include/python3.8 -c c/_cffi_backend.c -o build/temp.linux-armv7l-3.8/c/_cffi_backend.o
2022-09-25 13:39:40,322 ! c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory
2022-09-25 13:39:40,323 ! 15 | #include <ffi.h>
2022-09-25 13:39:40,324 ! |          ^~~~~~~
2022-09-25 13:39:40,325 ! compilation terminated.
2022-09-25 13:39:40,326 ! error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
2022-09-25 13:39:40,328 ! ----------------------------------------
2022-09-25 13:39:40,329 ! ERROR: Failed building wheel for cffi
```

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Unable to restart octoprint at all (no system menu is in the ui)

### Version of OctoPrint

1.8.2

### Operating system running OctoPrint

Armbian 21.08.1 Focal with Linux 5.10.60-sunxi

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[plugin_softwareupdate_console.log](https://github.com/OctoPrint/OctoPrint/files/9640706/plugin_softwareupdate_console.log)
[octoprint-systeminfo-20220925141725.zip](https://github.com/OctoPrint/OctoPrint/files/9640713/octoprint-systeminfo-20220925141725.zip)

derMart, 2022-09-25:
Installing libffi via `sudo apt install libffi-dev` got me a little further.
Is it an option to add some text to the readme, indicating that if no precompiled argon2-cffi-bindings are available that python dependency needs a compiler libffi (and potentially other system dependencies I am not aware of) to compile and install?

I started the second attempt to update by going to the settings menu then to Software Update. update log finished with:
```
2022-09-25 14:50:10,830 > Installing collected packages: pyparsing, pycparser, typing-extensions, pygments, packaging, deprecated, commonmark, cffi, rich, limits, argon2-cffi-bindings, passlib, Flask-Limiter, argon2-cffi, OctoPrint
2022-09-25 14:51:10,321 > Attempting uninstall: OctoPrint
2022-09-25 14:51:10,322 > Found existing installation: OctoPrint 1.8.2
2022-09-25 14:51:22,335 > Uninstalling OctoPrint-1.8.2:
2022-09-25 14:51:22,337 > Successfully uninstalled OctoPrint-1.8.2
2022-09-25 14:51:39,382 > Successfully installed Flask-Limiter-2.6.3 OctoPrint-1.8.3 argon2-cffi-21.3.0 argon2-cffi-bindings-21.2.0 cffi-1.15.1 commonmark-0.9.1 deprecated-1.2.13 limits-2.7.0 packaging-21.3 passlib-1.7.4 pycparser-2.21 pygments-2.13.0 pyparsing-3.0.9 rich-12.5.1 typing-extensions-4.3.0
```
But the updater UI popup did not finish, keeping the update message:
[image]

So I just reloaded the web page. But going to the settings still it is showing I am running OctoPrint 1.8.2 as if the update failed

derMart, 2022-09-25:
only after another attempt to update, the updater popup shows some output other than "Updating, please wait." and finishes:
[image]
Seems the updater UI has some issues...

cp2004, 2022-09-25:
> So I just reloaded the web page. But going to the settings still it is showing I am running OctoPrint 1.8.2 as if the update failed

You do have to restart OctoPrint once the update is installed, just reloading the page is not enough.

derMart, 2022-09-26:
> > So I just reloaded the web page. But going to the settings still it is showing I am running OctoPrint 1.8.2 as if the update failed
> 
> You do have to restart OctoPrint once the update is installed, just reloading the page is not enough.

Thanks, yes got that now. But I guess that the updater popup does not show any output even if the update finished successfully is a bug (which may be better placed in a follow up ticket...)

foosel, 2022-09-26:
> Is it an option to add some text to the readme, indicating that if no precompiled argon2-cffi-bindings are available that python dependency needs a compiler, libffi (and potentially other system dependencies I am not aware of) to compile and install?

I've added `libffi-dev` to the list of required packages in the RPi/Debian [setup guide](). I'll also adjust the README a bit (though something like "`apt install` these packages" won't land in there, the README - like OctoPrint - is platform agnostic.

> But I guess that the updater popup does not show any output even if the update finished successfully is a bug (which may be better placed in a follow up ticket...)

I've seen this before, but I can't remember right now what the cause was. It was something environment related if I remember correctly, possibly pip, but I'm currently sadly drawing a blank.

closed 2022-09-26 by derMart

derMart, 2022-09-26:
Ah, the link to RPi/Debian "setup guide" just links to this very ticket ;-) I would be very interested in reading the guide, which is very hard to find (haven't found it by looking at the README and its links)

foosel, 2022-09-26:
Fixed the link. But literally linked from the README ;)

derMart, 2022-09-26:
yea sorry, I did not recognized the (debian) in brackets ;-)