OctoPrint/OctoPrint #4653: WebSocket does not give feedback about invalid auth message

crysxd, 2022-09-26:
### The problem

The shortened session duration in 1.8.3 caused issues with existing clients which assumed a longer lifespan for a session. When the `/api/login` endpoint was used to create a session which is then used in the `auth` message of the web socket, there is no feedback from OctoPrint that the session was invalid. OctoPrint does not send any messages which is the only indication that the `auth` failed.

As a possible solution, the websocket should react in one of the following ways:

- Send a response `{"authSuccess": true}` or  `{"authSuccess": false}` in response to the `auth` message
- Do the above but also close the websocket in case of a failure (similar to what a 403 basically would do). This could help existing clients to automatically use the login endpoint again to open a new session.

I'm happy to supply a PR if an approach is agreed on.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.3

### Operating system running OctoPrint

/

### Printer model & used firmware incl. version

/

### Browser and version of browser, operating system running browser

/

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220926071645.zip](https://github.com/OctoPrint/OctoPrint/files/9643841/octoprint-systeminfo-20220926071645.zip)

Bristol-SuShuo, 2022-09-26:
Hi, May I ask you how do you get the log file of the terminal?
Thank you every much!

cp2004, 2022-09-26:
@Bristol-SuShuo That seems like a question better suited for the community forums or the discord server, as it's not related to this issue.

foosel, 2022-09-27:
Turns out this was actually a bug in disguise, as the handling of `reauthRequired(reason='stale')` messages was broken. Sending that in case of an unknown session during handling of `auth` certainly makes sense. Ready for 1.8.4.

crysxd, 2022-09-27:
Hope I didn't break that when adding the subscriptions :D Thanks!

commit be0d452a (2022-09-27): 🐛 Fix stale session reporting

Fix handling of `stale` reauthRequired in UI and
document it. Also send it in case of an `auth`
message with an unknown session.

Closes #4653