OctoPrint/OctoPrint #4654: Update to 1.8.4 on arm fails

nionio6915, 2022-09-29:
### The problem

Update to 1.8.4 from 1.8.1 fails on armbian.  

System logs do not clearly indicate what package is the problem. 
[octoprint.zip](https://github.com/OctoPrint/OctoPrint/files/9677792/octoprint.zip)

I have reviewed the docs at 
https://github.com/OctoPrint/OctoPrint/releases/tag/1.8.4
and 
https://community.octoprint.org/t/my-octoprint-update-fails-or-cant-be-started-at-all/37880
but do not recognoze my issue. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

armbian 

### Browser and version of browser, operating system running browser

google chrome & MS Edge

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

nionio6915, 2022-09-29:
bundle was included in the original zip file with the customarily requested files.

foosel, 2022-09-30:
You seem to be missing `libffi-dev`. Try installing that (`apt install libffi-dev`), then try to update again.

nionio6915, 2022-09-30:
that seems to have fixed the upgrade to 1.8.4. 

but now there are messages about failing to restart the server. i think that is a user permission issue. not sure if its related.

mcer12, 2022-10-01:
@nionio6915  Now I'm afraid to even try this

nionio6915, 2022-10-01:
i am pretty sure the server restart is a sudoers permission issue and unrelated.

cp2004, 2022-10-04:
It looks to me like this issue is solved, and so I will close it.

Please open a topic on the community forums or find the OctoPrint discord server for any further support questions you might have.

closed 2022-10-04 by cp2004