OctoPrint/OctoPrint #4655: Flask links in docs are broken, need fixing

foosel, 2022-09-30:
### Problem

The links to the Flask docs in the OctoPrint docs are broken. Most of them point to old releases for which the docs are no longer available on Flask's pages.

Example: https://docs.octoprint.org/en/master/plugins/mixins.html#blueprintplugin, "Blueprint mechanism" points to http://flask.pocoo.org/docs/0.10/blueprints/ which 404s.

### Solution 

Update all the broken links to working ones.

### Additional information

Mentioned by @JoveToo on [Discord](https://discord.com/channels/704958479194128507/705047010641838211/1025119342347694150).

DelfCo, 2022-09-30:
I'm used to working with Markdown and static site generators that aren't Sphinx, but I'd be willing to take this on as a learning exercise. I've found a number of pocoo.org and palletsprojects.com links in the docs directory that take me to the old pages, but with a banner at the top saying they're old and redirecting to current docs. I could update the links in these reStructuredText files to those current docs if you'd like.

DelfCo, 2022-10-03:
Two questions:
1. Most of this, I'm just updating old links to pocoo.org with new links to the latest versions, which all seem to be at palletsprojects.com now. Is that domain name change expected?

2.  I found this footnote in gettingstarted.rst:
In body text near the end of the topic: 
```
Linked here are the docs for Jinja 2.8.1, which OctoPrint still relies on for backwards compatibility reasons [#f3]_.
```

    And the footnote `[#F3]` says:
``` 
         Please always consult the Jinja documentation at `jinja.octoprint.org <http://jinja.octoprint.org>`_ instead of
         the current stable documentation available at Jinja's project page. The reason for that is that for backwards
         compatibility reasons OctoPrint currently sadly has to rely on an older version of Jinja. The documentation
         available at `jinja.octoprint.org <http://jinja.octoprint.org>`_ matches that older version.
```

Thing is, that URL just redirects to octoprint.org, not to any Jinja docs. Plus, it's an http: link, which causes my browser to complain about the certificate when it automatically tries to redirect to https://jinja.octoprint.org. And then if I tell it to ignore the certificate issue, it just ends up at the octoprint.org main page anyway.

It looks to me like the footnote needs to be removed and the body text modified, but I wanted to check: Is there still a dependency on Jinja 2.8.1?

DelfCo, 2022-10-03:
Also: some of the changes are to reference topic links in comments in .py files `types.py` and `files.py`. Should I put this against `maintenance` rather than `master`, since it's technically changing code? 

I'm in the process of setting up a dev environment according to your docs so I can do the tests and and check suite, as if it were a code change.

jneilliii, 2022-10-03:
That jinja link used to redirect properly to a copy of the old docs IIRC, probably something wrong on the server side there. Pretty sure the jinja version requirement is still valid.

cp2004, 2022-10-04:
OctoPrint got rid of the old Jinja dependency a couple of versions ago. Gina deleted jinja.octoprint.org recently, so the links need to be updated.

On master we are using Flask 2.1, on maintenance Flask 2.2 - there's no old Jinja version pins so linking to the current 3.x docs is sufficient, as that's what's being used these days. The current docs are hosted at palletsprojects.com, so that's expected as well.

For changes outside of the `docs` folder, I generally make them against `maintenance`, since as you say it is sort of changing code. Anything just in the `docs` folder can be directly on `master`, in my opinion at least 😉

foosel, 2022-10-04:
I could have sworn I had gotten rid of references to jinja.octoprint.org when I got rid of it, but apparently something slipped through 😅 So yes, please get rid of the footnote and that link, it's no longer needed.

As @cp2004 said, anything referring to jinja docs should just go against regular Jinja docs now. I think it should be fair game to directly link to Flask 2.2 docs so we don't need to update the docs for `maintenance` separately. 

Regarding changes outside of `docs` folders, as long as *only docstrings* are touched, that's also fine on `master` for me if it means we get updated docs faster. 

Also, I saw your question on Discord regarding some build issues and will jump over there next.

foosel, 2023-01-23:
Fixed by the above commit. Since source files needed to be touched, the fixes won't be live in the "latest" docs until 1.9 is released.

commit 94408fd2 (2023-01-23): 📝 Fix outdated jinja & flask doc links

Closes #4655
closed 2023-05-23 by foosel