OctoPrint/OctoPrint #4657: [Request] Extensible chart markings / labels

rfinnie, 2022-10-02:
### Is your feature request related to a problem? Please describe.

The new Start/Stop/etc markings on the chart (from #4382) are nice, but they are hardcoded to a handful specific events. It would be awesome if plugins could take advantage of them as well.  I maintain [a plugin](https://plugins.octoprint.org/plugins/bedcooldown/) which begins the bed cooldown shortly before the end of the print, and it would be a good addition to show on the chart when that cooldown begins.

### Describe the solution you'd like

Conceptual invocation:

``` python
from octoprint.events import Events

class BedCooldown(octoprint.plugin.EventHandlerPlugin):
    def on_event(self, event, payload):
        self._logger.info("Cooling down bed")
        self._printer.commands("M140 S0")
        if hasattr(Events, "CHART_MARKING"):
            self._event_bus.fire(
                Events.CHART_MARKING,
                {
                    "label": "Cooldown",
                    "color": "#ffc300",
                    "description": "The bed has begun cooling down",
                },
            )
```

Label and color are self-evident.  Description could be used for hover text.  `hasattr()` for backwards compatibility.

jneilliii, 2022-10-02:
wonder if you could call it directly via injected properties somehow. this is the relevant function to call: [add_marking](https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/printer/standard.py#L716-L722). something like `self._printer.add_marking("Cooldown")` not sure how the js side will handle it properly, if at all...

rfinnie, 2022-10-02:
Yeah, IMHO `_event_bus` is the "proper" place to put events like this, but extending `add_marking` is probably the practical way to do it.

BTW, as is, the JS *really* does not like arbitrary markings. Most of the UI just disappears as soon as a plugin calls `add_marking`. :laughing: 
[image]

cp2004, 2022-10-02:
I would agree, extending `add_marking` seems more logical to me. If it got some light documentation in the process as well of course.

rfinnie, 2022-10-03:
PR opened, I think future discussion can go in there: #4658

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel