OctoPrint/OctoPrint #4659: Backup Download Fails

Bachomp, 2022-10-03:
### The problem

Downloading of backups via the web interface results in a 404 error.

Files located on the file system have single quotes around them.

[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.4

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Prusa Mk3s

### Browser and version of browser, operating system running browser

MacOS 12.6. Vivaldi 5.4.2753.51 (Stable channel) (x86_64) 

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20221003113304.zip](https://github.com/OctoPrint/OctoPrint/files/9696754/octoprint-systeminfo-20221003113304.zip)

jneilliii, 2022-10-03:
based on the name, looks like it might be my backup scheduler plugin and is also creating hidden files similar to #4500

jneilliii, 2022-10-03:
not sure how, but am I using the backup helpers incorrectly? 

https://github.com/jneilliii/OctoPrint-BackupScheduler/blob/1081d996e3536c622e7eb6e665684c91b8dfa557/octoprint_backupscheduler/__init__.py#L158-L161

cp2004, 2022-10-03:
I suspect that's an issue with the new backup name validation (1.8.3 I think?), because it has a custom name?

https://github.com/OctoPrint/OctoPrint/blob/27a576e352a9b3d9abee89ab92cd02012c0e39dd/src/octoprint/plugins/backup/__init__.py#L1377

Bachomp, 2022-10-03:
> Were those backups created through the web interface, CLI, some third party plugin?

Actually, good point Gina, I believe the backup creation is triggered using the third party "Backup Scheduler Version: 0.1.0" on the startup event

foosel, 2022-10-04:
I think @cp2004 found the culprit. In my tightening down of API endpoints to prevent abuse, I didn't think of arbitrary backup names given on CLI or by third party plugins. So I guess that change needs rolling back or at least some nerfing.

@Bachomp just to make sure, downloading a backup created through the regular Backup plugin works? If so it's indeed just the filename sanity check.

foosel, 2022-10-04:
> not sure how, but am I using the backup helpers incorrectly?
> 
> https://github.com/jneilliii/OctoPrint-BackupScheduler/blob/1081d996e3536c622e7eb6e665684c91b8dfa557/octoprint_backupscheduler/__init__.py#L158-L161

I don't think you are, but something is still going on given stuff is single quoted. Or is that just what ls does here in case of spaces in the filename? 🤔

Bachomp, 2022-10-04:
> Just did a test and the manual native backup and download is working. The issue only appears to be related to @jneilliii plugin.

foosel, 2022-10-04:
> Or is that just what ls does here in case of spaces in the filename? 🤔

Indeed it is:

```
$ ls
'file with spaces.txt'   file_without_spaces.txt
```

So, nerfing the name check it is, since third party plugins and the CLI can choose arbitrary names.

foosel, 2022-10-05:
Fix ready for 1.8.5

thisiskeithb, 2022-10-17:
After updating to 1.8.5 on a few Pis here, clicking download on existing backups through the OctoPrint web gui now fail with a 404 error (previously worked in 1.8.4). I can see/still download the zips through SFTP from `/home/pi/.octoprint/data/backup/`, so the files are there:

```shell
------------------------------------------------------------------------------
OctoPrint version : 1.8.5
OctoPi version    : 0.18.0
------------------------------------------------------------------------------

pi@testpi:~ $ ls /home/pi/.octoprint/data/backup
Test-Pi-backup-20221017-105634.zip
```

I created new backups under 1.8.5 and those also fail to download with a 404 error, so this seems like it affects all backups.

jneilliii, 2022-10-17:
I have confirmed the same issue on one of my instances.

jneilliii, 2022-10-17:
I can confirm that staging/bugfix does allow me to download backups again.

closed 2022-10-18 by foosel

Bachomp, 2022-10-18:
Also confirmed downloads are now working again

commit b4d0ed30 (2022-10-05): 🐛 backup: fix file validation

Just going by name breaks backups
with arbitrary file names created
from CLI or the helper.

Instead of matching on the name, we
now check whether it's a zip and
contains a metadata.json.

Closes #4659
commit ac6240a3 (2022-10-17): 🐛 Fix tornado download endpoint

Something broke during the initial fix for #4659.

Fixes #4659