OctoPrint/OctoPrint #4661: Upgrading from 1.8.3 to 1.8.4 breaks Octoprint

feuerwasser85, 2022-10-04:
### The problem

Upgrading from Version 1.8.3 to 1.8.4 breaks the startup of Octoprint.
Not able to create a Systeminfo Bundle because after the update /home/pi/oprint/bin/octoprint is empty (0 byte file)
```
pi@ender5:~/oprint/bin $ ls -la octoprint
-rwxr-xr-x 1 pi pi 0 Oct  4 09:49 octoprint
```

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why

After the Upgrade /home/pi/oprint/bin/octoprint is empty (0 byte)

### Version of OctoPrint

1.8.4

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Creality Ender 5 (Plus) Marlin 2.1.1

### Browser and version of browser, operating system running browser

Windows 10, Chrome 106.0.5249.91

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

feuerwasser85, 2022-10-04:
I tried manually upgrade octoprint on octopi using this article: 
Didn't work out due to an old pip version. Tried to upgrade it:
```
pi@ender5:~/oprint/bin $ ~/oprint/bin/pip install -U octoprint
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: octoprint in /home/pi/oprint/lib/python3.7/site-packages (1.8.4)
WARNING: You are using pip version 20.3.3; however, version 22.2.2 is available.
You should consider upgrading via the '/home/pi/oprint/bin/python3 -m pip install --upgrade pip' command.
pi@ender5:~/oprint/bin $ /home/pi/oprint/bin/python3 -m pip install --upgrade pip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /home/pi/oprint/lib/python3.7/site-packages (20.3.3)
Collecting pip
  Downloading https://www.piwheels.org/simple/pip/pip-22.2.2-py3-none-any.whl (2.0 MB)
     |████████████████████████████████| 2.0 MB 3.5 MB/s
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.3.3
    Uninstalling pip-20.3.3:
      Successfully uninstalled pip-20.3.3
Successfully installed pip-22.2.2

```
```
Sadly the upgrade didn't do anything:
pi@ender5:~/oprint/bin $ ~/oprint/bin/pip install -U octoprint
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: octoprint in /home/pi/oprint/lib/python3.7/site-packages (1.8.4)
pi@ender5:~/oprint/bin $ ls -la octoprint
-rwxr-xr-x 1 pi pi 0 Oct  4 09:49 octoprint
```

foosel, 2022-10-04:
Given that we currently are at 17000 instances of 1.8.4 according to the anonymous usage tracking, 12000 of which came from 1.8.3 like you, I think we can rule out a general issue here and file system corruption is way more likely - something commonly discovered during updates. Make sure to always power down your Pi only after you've safely shut it down, make sure to make regular backups and also switch SD cards once in a while (those things wear out!).

You might be able to get things working again with [this FAQ entry](https://community.octoprint.org/t/how-to-reinstall-octoprint/36628), but no promises made.

feuerwasser85, 2022-10-04:
Good point, but both of my octopi instances (Ender 5 and Ender 5 Plus) react the same. On one i downgraded successfully to 1.8.3, on the other i took a backup and tried to force the upgrade. No success so far as it infinitly restarts now.

feuerwasser85, 2022-10-04:
Fixed it by throwing a big hammer on it:
`/home/pi/oprint/bin/python3 -m pip install --upgrade pip`
` ~/oprint/bin/pip install --force-reinstall octoprint --no-cache-dir`
Installing without upgrading pip was not possible.
Seems like there was something wrong on both instances in octopi. I can not think of any special custom setting i made. Everything is almost standard exept a fixed ip, dns and of course a custom password for the pi user.
The only problem i have now is a permission issue:
```pi@ender5plus:~ $ systemctl restart octoprint
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to restart 'octoprint.service'.
Authenticating as: ,,, (pi)
Password:
==== AUTHENTICATION COMPLETE ===```

foosel, 2022-10-04:
What version of OctoPi is this?

feuerwasser85, 2022-10-04:
0.18.0

feuerwasser85, 2022-10-04:
I was excited too soon... A reboot of the octopi server brings infinitly reboots.

foosel, 2022-10-04:
Upgraded several stock 0.18.0 to 1.8.4 without any need to update pip, so I have no idea what's up there. Also this makes me once again worry about file system corruption:

```
Authenticating as: ,,, (pi)
```

Honestly, run a file system check, something severely broken on OS level here.

feuerwasser85, 2022-10-04:
You're right, this makes no sense. Sadly both raspberry pis now restart infinitely so no access via ssh anymore. Good thin i have backups of octoprint config. Will switch SD cards.

feuerwasser85, 2022-10-04:
Can't get the newest octopi image to work anymore. I add the following lines at the bottom of config.txt and set up a static ip in /etc/dhcpcd.conf. After reboot the pi cant be pinged.
enable_uart=1
dtoverlay=disable-bt

foosel, 2022-10-04:
As that is clearly something that has nothing to do with OctoPrint core and I think we've also agreed on this being an issue related to something in your environment/your setup, I'm going to close this *bug* however since it isn't a bug and would like to ask you to move over to either the community forums or the discord for further support (way more people there that can give hints and help debug). Thank you!

closed 2022-10-04 by foosel

feuerwasser85, 2022-10-04:
Thanks for your support and kindness! You do a very good job with one of the best 3D printing software tools around.
I have figured it out and it was in fact my mistake. NEVER use Notepad to edit configs! Use Notepad++ instead. Opened the same files and there were extra characters -.-
Everything is working again.

foosel, 2022-10-04:
Aaaah, yeah, Notepad and Textmate or what it was called (Mac...) have caused issues for users in the past 😅

Personally I use VSCode for everything, with whitespace made visible, helps a LOT, but still I sometimes trip over line ending issues.