OctoPrint/OctoPrint #4662: [Request] Smart(er) pause/cancel

Gregor-Gregor, 2022-10-04:
### Is your feature request related to a problem? Please describe.

When printing and hitting the 'Pause' or 'Cancel' button the Octoprint flushes gcode buffer... then leaves the nozzle at the place it stopped. The hot nozzle melts the printout and oozes there. 

### Describe the solution you'd like

1. First of all it would be good to add Z-axis shift and moving of the printhead out of the printing area, the best to the back right corner. This would solve the issue described above.
2. For 'Pause' it would be good to add an option to execute pause _after the current layer is completed_. Why? Such a pause would be extraordinary tool for occasional drip or stringing removal, which unattended would dirty the hotend or otherwise disrupt the printoiut. At the moment I'm doing it _live_ trying to find a moment when printer is busy somewhere else. Having printhead stationary far far away would be much safer solution.

jneilliii, 2022-10-05:
That's what the gcode scripts feature is for. You can configure it any way you want. See example here for some pointers.

https://docs.octoprint.org/en/master/features/gcode_scripts.html#more-nifty-pause-and-resume

closed 2022-10-05 by jneilliii

Gregor-Gregor, 2022-10-05:
Aaah, thanks for this clue :)