OctoPrint/OctoPrint #4663: [Request] Allow plugins to estimate print progress in % and use that in UI/api/hooks etc

arekm, 2022-10-06:
### Is your feature request related to a problem? Please describe.

First what I consider a progress in %. Currently OctoPrint shows progress in % that is based on gcode file position. For users file position doesn't matter at all. They do care about % progress in relation to time.

Since OctoPrint uses not best source for % progress some plugins try to override that with better % estimation based on slicer (M73 command) estimations, previous history and other sources.

The problem now is plugins have no way to override % progress value in good way that's propagated to all progress users.

Plugins do workarounds like:

* override in javascript. Works for web UI but doesn't work for API (https://docs.octoprint.org/en/master/api/datamodel.html#sec-api-datamodel-jobs-progress ; btw. that API is quite as it provides filepos without entire file size) users. So web UI can get proper progress while API users (mobile app) will get different and invalid % progress values.

* override various places like _printer._stateMonitor._on_get_progress, util.getProgress(), _comm.getPrintProgress() and other hacks. These in theory work but gcodeviewer plugin *needs* file based progress to be passed to it (https://github.com/arekm/octopi_eta_override/issues/16). Changing above methods breaks gcodeviewer.

### Describe the solution you'd like

One solution: document util.getProgress() as "as close possible time progress estimate" and switch gcodeviewer plugin from getProgress to some other getFileProgress.  Plugins could override this with own estimates then.

Other solution: fully blown interface just like PrintTimeEstimator

on_print_progress() would also need to be extended or another on_print_time_progress() would be needed.

### Describe alternatives you've considered

Only these mentioned above. Any other that make sense?

jneilliii, 2022-10-06:
I think this is a good idea. I know plugins like PrintTimeGenius are doing the UI approach only, maybe something that could be added to the [octoprint-printer-estimation-factory ](https://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-printer-estimation-factory) hook would make sense here or adding another hook that plugins could override to return their own values for the API response directly.

Preclowski, 2022-11-03:
It may be extremely useful for Prusa owners! +1 on this