OctoPrint/OctoPrint #4664: Plugin not loading due to PIL import error (Docker image)

apfelnutzer, 2022-10-09:
### The problem

The plugin octoprint-webhooks is not loading anymore:
 
`ImportError: cannot import name '_imaging' from 'PIL' (/octoprint/plugins/lib/python3.8/site-packages/PIL/__init__.py)`

I use the octoprint/octoprint image with ID 32d643752f07 (latest as per today).

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Issue is about a 3rd party plugin not loading...

### Version of OctoPrint

1.8.4

### Operating system running OctoPrint

Raspberry Pi / Docker

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20221009110854.zip](https://github.com/OctoPrint/OctoPrint/files/9741385/octoprint-systeminfo-20221009110854.zip)

cp2004, 2022-10-09:
This is an issue with the 3rd party plugin, not OctoPrint itself. Please report it to the plugin author directly.

closed 2022-10-09 by cp2004

apfelnutzer, 2022-10-10:
Hm, I'm not sure this is related to the plugin, [but I raised an issue there](https://github.com/2blane/OctoPrint-Webhooks/issues/36).

apfelnutzer, 2022-10-10:
I thought it might be related to the plugin manager, but if 'pillow' is a plugin dependency you're probably right!