OctoPrint/OctoPrint #4667: [Request] Fire event to indicate when the folder watchdog adds a new file

smartin015, 2022-10-15:
### Is your feature request related to a problem? Please describe.

My plugin ([Continuous Print](https://github.com/smartin015/continuousprint)) currently relies on `Events.UPLOAD` to indicate when a new printable file is added to the file manager. It uses this as a signal that the user is adding work to OctoPrint (e.g. via drag-and-drop) and has configurable behavior to auto-add it to a queue of print items. 

However, users that configure directory watching for file import see different behavior - the [watchdog](https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/server/util/watchdog.py) implementation calls FileManager's `add_file()` function, which only calls `FILE_ADDED` and not `UPLOAD`.

### Describe the solution you'd like

I'd like the watchdog to fire either:

1. `Events.UPLOAD`, which admittedly stretches the term a bit but does describe an "upload" from the watched folder into the filemanager, or
2. A new event (`WATCHED_FILE_ADDED`?) that can be handled separately by my plugin.

### Describe alternatives you've considered

I know that the file manager fires `Events.FILE_ADDED`, but that event is also called whenever a file is moved/renamed and does not indicate a user's intent to push a file to OctoPrint for printing (they could just be moving files around, in which case they shouldn't see a bunch of files added to the queue).

Another option would be to extend the payload of `FILE_ADDED` to indicate whether this is a totally new file or just a move. I could use that to discriminate on the event and handle it accordingly.

### Additional context

I'm happy to provide a PR once a course of action is determined. Thanks!

jneilliii, 2022-10-15:
Why not use the [FileAdded](https://docs.octoprint.org/en/master/events/index.html#file-handling) event? I'm pretty sure that triggers on files added both ways.

foosel, 2022-10-26:
So, the [docs](https://docs.octoprint.org/en/master/events/index.html#file-handling) have this to say about the `Upload` event:

> Upload
>
> A file has been uploaded through the **[REST API](https://docs.octoprint.org/en/master/api/files.html#sec-api-fileops-uploadfile)**.

Thus, adding this event to a file addition through the watchdog would be a quite significant behaviour change, and most of the payload fields also don't make any sense for watchdog items.

But I also see the issue here with `FileAdded` not allowing to distinguish between a copy, a move and a new addition. @jneilliii is right in that this has historical reasons.

I'm not too keen on extending the meaning of `Upload` to also mean "somehow added by the watchdog". Actually, long term I'd rather get rid of it altogether than assign it new meaning 😬 Instead, we should think about how `FileAdded` is used. I think the clean answer here really might be an additional payload parameter, as already suggested by @smartin015.

Or maybe the answer is a completely new event fired for any kind of "something file like entered our library", on upload, on watchdog, on whatever else. 

I really don't feel happy with simply hijacking the `Upload` event for this, as it is very much "REST API Upload" specific, but for the sake of solving this pragmatically, I could see myself giving this a go, as long as we have at least discussed the option of the payload extension.

As far as I can see right now, the problem primarily arises because we also fire `FileAdded` on copy and move. If we were to mark copy and move event triggers with some kind of `operation` parameter that simply defaults to empty on regular calls to `file_added`, we should be fine, no?

jneilliii, 2022-10-26:
I'd be worried about potential infinite loop scenario possibly for those that may use upload events already and dropping files into watched folder, ie Google Drive Files plugin. Not completely sure that would happen.

foosel, 2022-10-26:
How so?

jneilliii, 2022-10-26:
now that I think about it, probably not. was thinking of a situation where if FileAdded event triggered something that then put another file into watched folder, which triggered another FileAdded event etc. My GoogleDriveFiles plugin uses the watched folder as an entry for new files being synched across so that other plugin's are able to process on FileAdded events as well as uploading files back up to Google Drive. I keep internal tracking list to prevent the loop myself in the plugin so files being added from watched folder are not then attempted to re-upload to Google Drive.

smartin015, 2022-10-30:
Hey, just checking in - I want to be sure everyone is in agreement on next steps.

Should I revert my PR and issue a new one that extends `FileAdded` with an `operation` parameter?

foosel, 2022-10-31:
From my side that would be the best approach. Maybe @jneilliii can also confirm, because I'm still not 100% sure I understood the problem he was seeing 😅

jneilliii, 2022-10-31:
I think that approach would be fine.

foosel, 2022-11-02:
So there we have it @smartin015, yes, please change your PR to one that extends `FileAdded` with an `operation` parameter instead. Or alternatively close it and open a new one, I'll leave that up to you :)

smartin015, 2022-11-15:
Friendly update - been sidetracked with other projects of late, but hoping to update my PR later this week.

smartin015, 2022-11-28:
Apologies for the delay - #4687 is finally open with the new `operation` parameter added.

smartin015, 2022-12-08:
Just checking in - anything I still need to do before #4687 is ready to review?

jneilliii, 2022-12-08:
No, Gina has just been a little under the weather lately and will soon be starting her annual Christmas vacation.

smartin015, 2022-12-08:
> No, Gina has just been a little under the weather lately and will soon be starting her annual Christmas vacation.

Got it - thanks. Both of those are more important :)

foosel, 2022-12-12:
That was a fast review, so I just did it quickly while wrapping up this year :)

closed 2023-05-23 by foosel