OctoPrint/OctoPrint #4669: Something crashed inside the serial connection loop

mattcalavera, 2022-10-21:
### The problem

What were you doing?
Printing an stl file via octoprint
Printed successfully until Octoprint Froze. I restarted server via SSH terminal. Octoprint ran as normal.
Pi Di dnot need to be rebooted.

Version of OctoPrint
1.8.6

Operating System running OctoPrint
Raspbian GNU/Linux 9 (stretch)

Printer model & used firmware incl. version
FLSUN SR

Browser and version of browser, operating system running browser
Chrome versione 106.0.5249.119 (Build ufficiale) (arm64)

EXTRACT FROM LOG: 
2022-10-21 21:27:45,704 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 4332, in _sendCommand
    if self._serial is None:
SystemError: unknown opcode

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 2515, in _monitor
    self._handle_ok()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3208, in _handle_ok
    self._continue_sending()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3429, in _continue_sending
    elif job_active and self._send_from_job():
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 4110, in _send_from_job
    f"fileline:{lineno}",
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 4418, in _sendCommand
    return enqueued_something
TypeError: 'cell' object is not callable
****

### Did the issue persist even in safe mode?

No, it did not persist (it's an issue with one of your installed plugins, don't report here)

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

OctoPrint 1.8.6 Python 3.7.3 OctoPi 0.18.0

### Printer model & used firmware incl. version

FLSUN SR

### Browser and version of browser, operating system running browser

Chrome versione 106.0.5249.119 (Build ufficiale) (arm64)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2022-10-22:
> OctoPi 0.18.0

> Operating System running OctoPrint
> Raspbian GNU/Linux 9 (stretch)

These two facts disagree - OctoPi 0.18 is based on Debian 10 (Buster), not Stretch. I can't verify which it is myself because there is no systeminfo bundle. It's no good posting extracts from the logs - we would like the **entire** thing.

mattcalavera, 2022-10-23:
Ok, here it is my systeminfo bundle.

I'm running in a [Raspberry Pi Zero 2 W] and once on 7/8 prints system hang and strop printing
[octoprint-systeminfo-20221023203336.zip](https://github.com/OctoPrint/OctoPrint/files/9847270/octoprint-systeminfo-20221023203336.zip)

foosel, 2022-10-26:
Unfortunately this system info bundle doesn't contain the reported error. Please reproduce (or wait until it reproduces on its own), then create and share the bundle right away.

From the looks of the log excerpts above however, something gets seriously messed up in the Python process. "Unknown opcode" sounds like it either has some corrupt code its dealing with (ever shut off the Pis without a proper shutdown procedure first?) or some other fishy thing is going on. It's certainly not something within the control of OctoPrint, and the code location also doesn't make a lot of sense here.

You might want to try if a [reinstall](https://community.octoprint.org/t/how-to-reinstall-octoprint/36628) fixes it.

cp2004, 2022-12-11:
@mattcalavera were you able to test if a reinstall fixed the issue? I would agree with Gina's assessment that the crash was not something OctoPrint could control, but it would be interesting to know if you were able to reproduce the issue again.

mattcalavera, 2022-12-11:
Charlie,Sorry but i can’t, i migrate to klipper MFing. Mattia Fiorini

closed 2022-12-12 by foosel