OctoPrint/OctoPrint #4670: Cannot grant access / give permissions to guests

loelkes, 2022-10-22:
### The problem

I want to grant guests access to API resources, also from plugins. So I went to the `Access Control` page in the Settings and gave the permissions to guests. Maybe this is already no intended?

* the job API. Source code says permissions for `STATUS` are required. Granting access for guests in the UI and calling `/api/job` returns `{"error":"You don't have the permission to access the requested resource. It is either read-protected or not readable by the server."}`
* the Announcement Plugin. Granting read access for guests in the UI and calling `/plugin/announcements/channels` returns `{"error":{"dangerous":false,"default_groups":["admins"],"description":"Allows to read announcements","key":"PLUGIN_ANNOUNCEMENTS_READ","name":"Announcement Plugin: Read announcements","needs":{"role":["plugin_announcements_read"]},"plugin":"announcements"}}`

Why are there different error messages (?) for the same issue (it seems?).

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

MacOS

### Printer model & used firmware incl. version

Virtual

### Browser and version of browser, operating system running browser

macOS 12.6 / Chrome 106.0.5249.119 / curl 7.79.1

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20221022205926.zip](https://github.com/OctoPrint/OctoPrint/files/9845096/octoprint-systeminfo-20221022205926.zip)

cp2004, 2022-10-22:
Guest users still require a session cookie to work, which can be obtained from a passive login POST to `/api/login` first to effectively login as a guest, get a session cookie and then you can access further resources on the server. We can debate whether this is correct or not but that is how it works (and as far as I know always has done). No password is required to get a session cookie, but it must still be done.

The different error message for the announcements plugin's endpoint is interesting, I've not seen that before. I can reproduce it, and will look at why that might be.

loelkes, 2022-10-22:
I did not know that. At no point I find a hint about this in the docs.

1. BlueprintPlugin / is_blueprint_protected()
> Whether a login session by a registered user is needed to access the blueprint’s endpoints. Requiring a session is the default. Note that this only restricts access to the blueprint’s dynamic methods, static files are always accessible.
>
> If you want your blueprint’s endpoints to have specific permissions, return False for this and do your permissions checks explicitly.

The first paragraph implicitly says that if I return `False`, a session if not required by default anymore. Which is true, as long as I don't do the mentioned explicit permission checks.

2. Access control

> The predefined “Guests” group can be used to configure default permissions of anonymous users, that is those who have not logged in. By default, no permissions are granted to these users.

It does not mention the necessity for a session cookie, but it also does not deny it.

I don't think requiring a session cookie for unauthenticated/anonymous requests against an API is reasonable. Is this up for discussion somewhere or can I start questioning it?

loelkes, 2022-10-24:
Yes, I've only checked the REST API. Please let me know if or how I can help.

foosel, 2022-10-26:
Can confirm this and it is definitely a bug. Now need to figure out how that happened.

foosel, 2022-10-27:
Has been around since at least 1.7.3, so not a regression. Looking at the code, this has been broken for a *long* time indeed. On anonymous calls to the API without an API key, an empty `AnonymousIdentity` was set. That didn't get assigned the permissions configured for the anonymous user and that causes this behaviour. In the browser, Flask creates a session context, and that *does* get the correct anonymous permissions assigned, and thus there it works.

So, long time bug that no one noticed up until now. No regression, so target version for a fix is 1.9.0.

foosel, 2022-10-27:
Fix ready for 1.9.0. Also thanks for the error message heads-up, fixed that as well.

commit 7147062d (2022-10-27): 🐛 Fix permissions of anonymous API access

Was running with a full blank AnonymousIdentity,
even if guest users had permissions configured.

Closes #4670
closed 2023-05-23 by foosel