OctoPrint/OctoPrint #4671: Serial resending fails due to trying resending wrong line number (N value) not being in history buffer anymore

Quas7, 2022-10-26:
### The problem

Issue happend with fresh update to 1.5.2 now two times in a row for same model but different slicing-gcode.

Printing stops with "hotend idle timeout" message after some longer print time.
The root cause seems to be a failing resend of a missed G1 command.
Resend fails because it loses 100k lines in between (missing "1" first digit).
Also, the Seriallog-Errormessage "ErroNchecksum mism**1**" seems to be corrupted and it ends with a rather suspicious "1".

Note: resends happen only occasionally with 75 / 140k (0%) at 250kbaud. 
Likely only resends above 100k lines fail - both prints failed there for now.

```
Send: N**134497** G1 X18.64 Y-54.548 E10.8357*72
Recv: ErrorNchecksum mism**1**
Recv: ch, Last Line: **134496**
Recv: Resend: **034497**
Should resend line 34497 but no sufficient history is available, can't resend
```

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

heater timeout plugin has been disabled. Communication issue is likely root cause. Plugins with high CPU load possible an issue but not tested yet.

### Version of OctoPrint

1.5.2

### Operating system running OctoPrint

Raspbian

### Printer model & used firmware incl. version

Delta printer Flsun Q5 with Marlin 2.6.1 (stable)

### Browser and version of browser, operating system running browser

Chrome 106.0.5249.119-64bit, Windows 10-64bit

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

```
Send: N134494 G1 X20.739 Y-54.274 E10.76169*65
Recv: ok N134494 P13 B31
Send: N134495 G1 X20.515 Y-54.477 E10.77194*74
Recv: ok N134495 P14 B31
Send: N134496 G1 X20.229 Y-54.548 E10.7819*119
Recv: ok N134496 P16 B31
Send: N134497 G1 X18.64 Y-54.548 E10.8357*72
Recv:  T:235.48 /236.00 B:61.42 /60.00 @:79 B@:0
Recv: ErrorNchecksum mism1
Recv: ch, Last Line: 134496
Recv: Resend: 034497
Should resend line 34497 but no sufficient history is available, can't resend
Changing monitoring state from "Printing" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
Closing down send loop
```

gcode input section (nothing special):
```
G1 X20.829 Y-54.051 E10.75357
G1 X20.739 Y-54.274 E10.76169
G1 X20.515 Y-54.477 E10.77194
G1 X20.229 Y-54.548 E10.7819
G1 X18.64 Y-54.548 E10.8357
G1 X18.64 Y-58.8 E10.97962
G1 X27.375 Y-58.8 E11.27532
```
[octoprint-systeminfo-20221026115640.zip](https://github.com/OctoPrint/OctoPrint/files/9868778/octoprint-systeminfo-20221026115640.zip)

cp2004, 2022-10-27:
I don't think there's anything that OctoPrint itself could do differently here - you've run into some communication issues, and they are unrecoverable. The line number that has been requested is wrong, potentially due to the same serial communication issues. Perhaps Marlin is sending the correct things, but somewhere along the line they are corrupted. It is hard to know for sure.

In an ideal world, you would aim for 0 resends *at all*. For me, it happening 75 times is too many and is indicative of some kind of small communication issue. Yes it's not necessarily critical, but it will still cause pauses in the print as the communication recovers.

Also - you don't have OctoPrint 1.5.2 at all, nor did you just upgrade to it. Your log shows you upgraded from 1.8.4 to 1.8.6.

foosel, 2022-10-27:
```
2022-10-25 08:21:46,991 - octoprint.plugins.pi_support - WARNING - This Raspberry Pi is reporting problems that might lead to bad performance or errors caused by overheating or insufficient power.
!!! UNDERVOLTAGE REPORTED !!! Make sure that the power supply and power cable are capable of supplying enough voltage and current to your Pi.
```

And then there's also this, which might be a factor here as well.

I agree with the assessment by @cp2004, there's nothing that OctoPrint can do differently here. If there comes garbage in on serial, it cannot reliably communicate with the printer. It can't magically turn garbage into working messages. It could be that this happens on the line, it could be this happens in the serial port driver, but the end result is, garbage arrives on the serial line for OctoPrint to read, and that's not something we can fix.

Please get in touch on the forums or the Discord if you need further assistance in debugging your communication issues. Closing this since it is not actionable.

closed 2022-10-27 by foosel