OctoPrint/OctoPrint #4674: [Request] Just-in-time gcode preview

richfelker, 2022-10-28:
### Is your feature request related to a problem? Please describe.

GCode Viewer is unusable on large models. The volume of data overloads the browser and makes the tab crash and/or slow to a crawl.

### Describe the solution you'd like

Rather than loading and analyzing the entire gcode file, only index the start of each layer at load time. Then, using that index, load, analyze, and render a particular layer only when navidating to it. Cache up to (a configurable?) N recent layers for quicker navigation and for showing previous/next layer if selected.

### Describe alternatives you've considered

The only alternatives seem to be not using Octoprint for gcode preview or only using it from a very-high-end computer (and especially not mobile device).

cp2004, 2022-10-28:
Ready for 1.9.0, @JoveToo has implemented some compression and asynchronous download & processing of the gcode files. This makes the memory usage reduced to around 25% of what it was (at the expense of a longer load time) - you can view much much larger models with the gcode viewer in 1.8.x and especially 1.9.x (when released). This may not be as relevant after that. For a 60mb model, the gcode viewer only used 100mb of memory for me. It took a long time to load with compression but it did.

I think a JIT gcodeviewer could be a plugin, otherwise we will work with optimizating what we have already.

JoveToo, 2022-10-28:
You gotta have patience but in Chromium, I can load a 519MB gcode file without issue on the maintenance branch (future 1.9.0). I haven't tried larger files.

FYI, the limit is the memory size that Chromium allows for a single webpage to use, not the power of your computer.

richfelker, 2022-10-29:
I use Firefox not Chrom{e,ium}. Above around 25 MB it's slow and before long, unusable. Gets unusable even quicker on mobile (also Firefox; I don't use any Chromium-based browsers).

JoveToo, 2022-10-29:
@richfelker It should be a lot better on 1.8.x, but Firefox has less restrictions than chromium based browsers when it comes to memory. I do not use chromium based browsers either, Firefox works fine here. What Octoprint version are you on?

JoveToo, 2022-10-29:
I just tested maintenance branch on my phone with Firefox Focus and it parses the 519MB file fine. It is slow though, it takes several minutes to parse the file but it works and I can scroll through the layers just fine.