OctoPrint/OctoPrint #4675: Doesn't support usernames and passwords in webcam URLs

gitcnd, 2022-10-29:
### The problem

The settings page lets us specify our webcam stream and snapshot URLs, e.g:
http://myserver.local:8001?action=stream

It errors out if authentication is included, which is wrong.  e.g:-
http://myuser:mypasswd@myserver.local:8001?action=stream

It also fails to let us specify cam usernames and passwords by other ways

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

not relevant

### Version of OctoPrint

OctoPrint 1.7.3Python 3.6.8

### Operating system running OctoPrint

Linux (Centos 8)

### Printer model & used firmware incl. version

n/a

### Browser and version of browser, operating system running browser

Chrome on Macbook

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

no checklist files needed for this bug

gitcnd, 2022-10-29:
Not sure how this bug could even exist - the standard commands for working with URLs form the commandline do work fine, so something internally must be incorrectly parsing or processing the URLs ?

e.g. - loading from bash works fine:-

[root@hpz Err:1 octoprint]# curl  'http://octo:**@hpz.local:8001?action=snapshot' |wc     
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  297k    0  297k    0     0  8262k      0 --:--:-- --:--:-- --:--:-- 8498k
   1365    6059  304604
[root@hpz octoprint]# curl -i -s 'http://octo:**@hpz.local:8001?action=snapshot' 
HTTP/1.0 200 OK
Access-Control-Allow-Origin: *
Connection: close
Server: MJPG-Streamer/0.2
Cache-Control: no-store, no-cache, must-revalidate, pre-check=0, post-check=0, max-age=0
Pragma: no-cache
Expires: Mon, 3 Jan 2000 12:34:56 GMT
Content-type: image/jpeg
X-Timestamp: 5488.521990

[root@hpz Err:1 octoprint]# 

To actually get it to fail, something has to manually remove the username and password *from* the URL - whatever is doing that inside octoprint should stop doing that?:-

[root@hpz Err:1 octoprint]# curl -i -s 'http://hpz.local:8001?action=snapshot'               
HTTP/1.0 401 Unauthorized
Content-type: text/plain
Connection: close
Server: MJPG-Streamer/0.2
Cache-Control: no-store, no-cache, must-revalidate, pre-check=0, post-check=0, max-age=0
Pragma: no-cache
Expires: Mon, 3 Jan 2000 12:34:56 GMT
WWW-Authenticate: Basic realm="MJPG-Streamer"

401: Not Authenticated!
username and password do not match to configuration

cp2004, 2022-10-29:
For webcam streams, OctoPrint just embeds the URL in an <img> tag, it is up to your browser for the rest. Pretty much all (as far as I know all) browsers will block the use of usernames & passwords embedded in the URL like this. So it isn't an OctoPrint bug, it is a security feature of your browser.

For the snapshots, OctoPrint doesn't use `curl` to download the images, this is all done internally with the Python code. It is not a bug, but it is just not supported without extra legwork. Since authentication comes in many different shapes and sizes, adding support for it to OctoPrint's core is waiting until the webcam system is made to support plugins, which could be made to support any kind of authentication there might be on a webcam stream. I have a feeling there was already an issue/PR somewhere that discussed this but I can't find it now. Webcam plugins are targeted to land in 1.9.x. To update your install, you will need a newer version of Python - since 1.8.x, OctoPrint has only supported 3.7+.

closed 2022-10-29 by cp2004

cp2004, 2022-10-29:
Having said that final part about the snapshots, I just tested and it looks like username:password@URL should work there. Nothing is stripped out of the URL, and it seems to authenticate correctly. I don't currently have a webcam setup to test it, but used https://jigsaw.w3.org/HTTP/ to test (the basic auth page) and ended up with OctoPrint correctly downloading that page several times. Obviously it's not an image and so the timelapse doesn't *work*.

gitcnd, 2022-10-29:
Interesting!  I'll dig into why it thinks it failed on my box (I was using the "test" option in the UI - maybe it will just work after I reboot).

I notice on Chrome/Windows at least, that the following javascript in an HTML page works fine:-

var url='http://octo:**@hpz.local:8001?action=stream'
var myWindow = window.open("", "MsgWindow");
myWindow.location=url;

and also that thereafter, the browser auto-remembers the auth for the rest of the session - so while username/password might not work in embedded HTML elements directly, they can be made to work always in a popup/new-tab, and always indirectly (embedded in octoprint UI without needing the credentials in that URL) after a popup has happened.

It's a pity the original webcam work was done by someone who doesn't care about security or privacy - it really would have been trivial to make this all work properly without putting anyone at risk... that's a major contributor to why there's so much crime on the internet - nobody cares, nobody fixes anything, and nobody has skin tough enough to take deserved criticism when this is pointed out to them... shoot-the-messenger is the immediate reaction every time  :-(

jneilliii, 2022-10-30:
> It's a pity the original webcam work was done by someone who doesn't care about security or privacy - it really would have been trivial to make this all work properly without putting anyone at [...]

What are you trying to get at with this comment? The webcam stuff is being actively worked on to make it more open and possible to support your exact circumstance. I don't think anyone tried to "shoot the messenger".

cp2004, 2022-10-30:
Yeah I don't get this comment either. We are blocked in the browser from using embedded credentials in the stream URL, and on the snapshot side there is a PR adding support for webcam plugins, which could definitely add support for different authentication types. What should we be doing differently?