OctoPrint/OctoPrint #4676: Hot panic crash

mullcom, 2022-10-30:
### The problem

Hot run away panic.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

latest

### Operating system running OctoPrint

0.18.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

https://photos.app.goo.gl/xQ4MbaKzNhxTTKZA8

Send: N79046 G1 X164.095 Y85.931 F12000*73
Recv: ok
Send: N79047 G1 Z2.2 F720*34
Recv: ok
Send: N79048 G1 E1.3 F2100*6
Recv: ok
Send: N79049 G1 F3600*72
Recv: ok
Send: N79050 G1 X163.508 Y85.344 E.01687*102
Recv: ok
Send: N79051 G1 X163.294 Y85.276 E.00456*106
Recv: ok
Send: N79052 G1 X162.983 Y85.131 E.00697*106
Recv: ok
Send: N79053 G1 X162.717 Y84.913 E.00698*110
Recv: ok
Send: N79054 G1 X162.48 Y84.679 E.00677*86
Recv: LCD status changed
Recv: Error: HOTEND THERMAL RUNAWAY
Changing monitoring state from "Printing" to "Error"
Send: M112
Send: N79055 M112*47
Send: N79056 M104 T0 S0*44
Send: N79057 M104 T1 S0*44
Send: N79058 M104 T2 S0*32
Send: N79059 M104 T3 S0*32
Send: N79060 M104 T4 S0*45
Send: N79061 M140 S0*108
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

mullcom, 2022-10-30:
[octoprint-systeminfo-20221030121753.zip](https://github.com/OctoPrint/OctoPrint/files/9895781/octoprint-systeminfo-20221030121753.zip)

jneilliii, 2022-10-30:
> Recv: Error: HOTEND THERMAL RUNAWAY

This is an error with your printer and not an issue with OctoPrint. You should troubleshoot what is causing the thermal runaway error. You could ask on the [Community Forum](https://community.octoprint.org/) or in Discord in the `support-printer` channel for help.

closed 2022-10-30 by jneilliii

mullcom, 2022-10-30:
thx. i dident know if it was printer or not. temp was good so run away was wrong ditected from unit.