OctoPrint/OctoPrint #4677: Octoprint not sending heater commands to printer

pblawler, 2022-10-30:
### The problem

I run a print and everything works as expected.

I try to run the print a second time and octoprint shows that it sent the target Temps to the printer but the printer does not heat up and errors out as thermal run away.

If I restart octoprint after each print everything works as expected.   If I do not restart octoprint the heaters do not energize even though it shows the target temp was sent.  I have verified the voltage to the mosfets and everything is working properly on the printer side.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

Octopi

### Printer model & used firmware incl. version

Custom, Marlin 1.1.9

### Browser and version of browser, operating system running browser

Chrome 106.0.5249.119

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Log files

https://www.4shared.com/zip/gZ2R4nMDfa/octoprint-logs.html

pblawler, 2022-10-30:
I included a link to the log files in the original submission. [Log files](https://www.4shared.com/zip/gZ2R4nMDfa/octoprint-logs.html)

pblawler, 2022-10-30:
Please advise how to directly upload the log files

jneilliii, 2022-10-30:
Your log doesn't indicate being run in safe mode best I can tell. Also in your log, notice a couple of things like 

> 2022-10-28 13:00:51,974 - octoprint.util.comm - WARNING - Not sending "M141 S60.0", printer profile has no heated chamber. Either configure a heated chamber or remove chamber commands from your GCODE.

do you have a heated chamber? if not you should maybe remove that from your slicer's settings. 

you will probably also need to enable serial logging in the printer > serial connection settings of OctoPrint to diagnose better.

foosel, 2022-10-31:
Please don't just provide a single log file. Enable `serial.log` as described by @jneilliii, reproduce your error, then create & provide a systeminfo bundle: [What is a Systeminfo Bundle and how can I obtain one?](https://community.octoprint.org/docs?topic=29887)

cp2004, 2022-12-11:
@pblawler did you have a chance to create a serial.log to share (as part of the systeminfo bundle) with us to try and find the issue?

closed 2023-01-24 by foosel