OctoPrint/OctoPrint #4680: Bug in file list after dropping or uploading a file - top lines hidden

Clumpton, 2022-11-07:
### The problem

When you upload or drag/drop a file onto the files panel of the web interface and the list is sorted by upload date the file goes to the head of the list and flashes yellow. However the top half of the entry is hidden and only the last two lines of the file details are visible. The slider is shifted down slightly. If the list is empty or short (i.e without a slider), the problem does not arise

Behaviour is the same in Firefox, Chrome and Opera.

Behaviour persists and is aggravated in Safe Mode only one last line is visible.

[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

     OctoPrint 1.8.6     Python 3.7.3     OctoPi 0.18.0

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Ender 3V2 + mriscoc Ender3V2-422-BLT-20221002

### Browser and version of browser, operating system running browser

Firefox, Opera, Chrome (all up to date), Windows 10 Pro 21H2

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Javascript console remains empty.

[octoprint-systeminfo-20221107145559.zip](https://github.com/OctoPrint/OctoPrint/files/9953681/octoprint-systeminfo-20221107145559.zip)

Video of it happening:
[scrnli_11_7_2022_6-14-17 PM.webm](https://user-images.githubusercontent.com/59048696/200372998-bbb4d705-8728-41ad-8d73-b7e45c5569b5.webm)

foosel, 2022-11-07:
Just confirmed on 1.9.0dev as well.

foosel, 2022-11-07:
Fix ready for 1.9.0.

cp2004, 2022-11-07:
I tried to get to the bottom of this one when adding multiple upload support, didn't think it would be that simple and at the time I couldn't reliably reproduce it 👍

commit 792de0b4 (2022-11-07): 🐛 Fix file list scroll position on upload

File list entries had the wrong offset parent
due to a missing `position: relative`.

Closes #4680
closed 2023-05-23 by foosel