OctoPrint/OctoPrint #4681: [Request] "File Already Exists" dialogue - Add information on old/new files

Clumpton, 2022-11-07:
### Is your feature request related to a problem? Please describe.

When you upload a gcode file to Octoprint and a file of the same name is already present a "File Already Exists" dialogue is show offering  to Cancel, Rename or Overwrite it. (see screenshot)
[image]
There is no indication of the age, size or upload date/time of the pre-existing file, this could lead to confusion or loss of a precious gcode file.

### Describe the solution you'd like

The "File Already Exists" dialogue should show additional file information to help decide whether it is safe to overwrite.
For example:
**Existing file foobar.gcode:**
- Size 250 kb
- Date 2021-12-25 15:35:20
- Uploaded 3 weeks ago

**New file foobar.gcode:**
- Size 721 kb
- Date 2022-1107 18:30

### Describe alternatives you've considered

I've considered being more careful and remembering old files, but I'm a wrinkly and my memory fails me :-)

max246, 2023-01-30:
I have made a PR for this https://github.com/OctoPrint/OctoPrint/pull/4720

Clumpton, 2023-01-31:
> I have made a PR for this #4720

Thanks for that!

foosel, 2023-03-08:
Implemented by #4721, will be in 1.10.0

Clumpton, 2023-03-08:
Thanks for that 

⁣Get TypeApp for Android ​

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel

Clumpton, 2024-05-01:
Nice implementation, thanks again.