OctoPrint/OctoPrint #4682: [Request] Warn users about missing `wheel` package

cp2004, 2022-11-09:
### Is your feature request related to a problem? Please describe.

A new warning has popped up in pip:

https://github.com/pypa/pip/issues/8559

Basically every plugin uses setup.py (and there's work to be done to switch that to pyproject.toml at some point here), and while the vast majority probably have the wheel package, maybe some don't. To avoid potential issues installing plugins/updates in the future, I wonder if a warning could be added to the SW update page or plugin manager gently mentioning that wheel is not installed. I don't think an intrusive warning is necessary here just yet, but it might preempt some issues?

### Describe the solution you'd like

err... I got carried away in the previous box.

### Describe alternatives you've considered

Suffering with support questions?

### Additional context

See linked issue, and also #4552 for the pyproject.toml stuff.

foosel, 2022-11-09:
Makes sense. Also reading through the linked pip issue, isolated builds will become the norm, so #4552 just got a huge priority boost.

foosel, 2022-11-10:
Hm... should we maybe just make `wheel` part of OctoPrint's requirements? Then it should get installed automatically if it's currently missing for whatever reason.

cp2004, 2022-11-10:
That does make more sense than a warning, yeah!

foosel, 2022-11-14:
`wheel` is now part of OctoPrint's requirements. For now without a version pin.

closed 2023-05-23 by foosel