OctoPrint/OctoPrint #4683: Octoprint "drop to upload file" when switching to firefox window

mashuptwice, 2022-11-09:
### The problem

1. open octoprint in firefox
2. leave tab open and switch to a different window
3. switch back to firefox
4. octoprint opens "drop to upload file" without a file dragged

[image]

https://user-images.githubusercontent.com/54219098/200918904-5c603cce-0ac9-4c14-8b18-5b4b9f6ed37c.mp4

**Notes:**

The issue does not occur if the tab was not visible and I switch to octoprint from within firefox. 

It might be worth noting that I am using octoprint on a thinkpad convertible, I could imagine the touchscreen or wacom AES being misinterpreted. 

It happens regardless of the mouse pointer and previous touch or stylus position. 

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I have a print running currently, will edit the bug report later.

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

Raspberry 3B+, Debian/Raspberry OS kernel 5.10.103-v7+, used Octopi image

### Printer model & used firmware incl. version

MKS Tinybee

### Browser and version of browser, operating system running browser

Mozilla Firefox 106.0.5, 6.0.7-arch1-1 KDE Wayland, plasmashell 5.26.3

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Does not occur when developer tools / javascript console is open.

[octoprint-systeminfo-20221109201336.zip](https://github.com/OctoPrint/OctoPrint/files/9974401/octoprint-systeminfo-20221109201336.zip)

jneilliii, 2022-11-09:
are you using keyboard shortcuts for switching applications, wasn't clear from the video.

jneilliii, 2022-11-09:
I just tested firefox 106.0.5 (64-bit) on windows and cannot reproduce this issue. tried both switching using the task bar and shortcut keys. I do know that if you have text selected and move the mouse on the page it will also show that drop overlay, so I'm wondering if that's what's happening here for you with the waacom, etc.?

mashuptwice, 2022-11-09:
@jneilliii it happens both with with keyboard shortcuts, as well as when clicking on the task bar. 

What do you mean by text selected? Like select and drag it into the open window, just selected or selected and copied to the clipboard?

mashuptwice, 2022-11-09:
Welp, I've just experienced the same on google search results. This might be an upstream thing then!

Tested Mozilla Firefox 106.0.3, the issue does not occur in this version. 

Upgraded to Mozilla Firefox 106.0.5 again, the issue does not occur! 

Closing this as not reproducible.

mashuptwice, 2022-11-11:
I'll reopen this issue, as it keeps happening. Downgrading to different Firefox versions did not help.

foosel, 2022-11-14:
Cannot reproduce this either. I use Firefox as my main browser myself. It might be a touchscreen issue, however I don't have access to one anymore. So that's going to need someone to look at who can reproduce this I fear.

mashuptwice, 2022-11-19:
Narrowed it down, it is an [upstream KDE bug.](https://bugs.kde.org/show_bug.cgi?id=446111)

closed 2022-11-19 by mashuptwice