OctoPrint/OctoPrint #4684: Chrome displays "Out of memory" -error when using cam

asdasddffgthhjtreresf, 2022-11-11:
### The problem

Latest release of octoprint shows Aw, Snap! Something went wrong while displaying this webpage. Error code: Our of memory
This happens when using Control -tab with cam. 

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I have long print going on so can't test safe mode.

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

The ready-to-go Raspberry Pi image with OctoPrint Version 0.18.0

### Browser and version of browser, operating system running browser

Chrome Version 107.0.5304.107 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20221112015323.zip](https://github.com/OctoPrint/OctoPrint/files/9993688/octoprint-systeminfo-20221112015323.zip)

foosel, 2022-11-12:
> I have long print going on so can't test safe mode.

Cool, then please test in safe mode once that print is done and then report back here.

cp2004, 2022-12-11:
@asdasddffgthhjtreresf did you ever get a chance to test this again? Are you using the standard mjpg streaming, or experimental HLS/WebRTC?

ReverendRhyme, 2023-01-12:
My Octo is also unusable all of a sudden while printing. If I'm not printing it runs great.

You cannot navigate through the interface without massive lag. 

Eventually resulting in an "out of memory" error. This happens in Chrome and Edge. Both are using the latest version.

cp2004, 2023-01-12:
@ReverendRhyme, are you perhaps able to pick up the unanswered question here, of testing it in safe mode? And this error only happens when the webcam is being shown, yes?

ReverendRhyme, 2023-01-12:
> 
I'll see If I can reproduce in Safemode once my print is done. It only crashes out while printing with the cam shown. I can have the camera view online while no print is going, and everything works as you would expect.

ReverendRhyme, 2023-01-12:
Well, everything seems ok in Safemode. So one of the plugins must be responsible.

cp2004, 2023-01-15:
That does make it most likely a plugin issue. If you find out which plugin it is, please let us know in case someone else comes across this, and it can be reported to the plugin author as an issue. Unless @asdasddffgthhjtreresf ever replies, this issue can probably be closed soon.

foosel, 2023-01-26:
Closing due to lack of response.

closed 2023-01-26 by foosel

3DStreetFurniture, 2023-07-04:
I get the same and can test this for you @foosel 
[image]