OctoPrint/OctoPrint #4686: Printing State sub-window doesn't update if identically named different file uploaded

frollard, 2022-11-27:
### The problem

[image]

I haven't attempted to replicate as I only need one of this part, but I believe it has a simple cause:
Only Plugin of note: Arc Welder
Uploading a duplicate file name does not update the print state file information on the new file when using arc welder.

1. Upload a file (aw converts it to .aw.gcode)
2. Start printing the aw.gcode file
3. Realize problem with print during bed heating, cancel print.
4. Edit original model, re-slice, re-upload with identical file name, 20% bigger file.
5. Use file duplicate tool to 'overwrite'
6. Arc Welder converts new file to .aw.gcode, automatically overwriting
7. Select the new identically named file.aw.gcode in the file browser, and use the print button in the file selection screen to start the print.
8. Print new file successfully, except the Print time left becomes "-", and the print progress bar proceeds past 100% up to (in this case about 120%). File data size printed (as in screenshot) shows 460KB/401.9KB, however the file printed is 518KB.
[octoprint-systeminfo-20221127055304.zip](https://github.com/OctoPrint/OctoPrint/files/10097624/octoprint-systeminfo-20221127055304.zip)

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

On a time constraint, cannot test - I believe it's unlikely AW caused the issue since the print command was sent from the file modal sub-button.

### Version of OctoPrint

OctoPrint 1.8.2 Python 3.7.3 OctoPi 0.18.0

### Operating system running OctoPrint

OctoPi 0.18.0

### Browser and version of browser, operating system running browser

Chrome Version 107.0.5304.107 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

No errors in javascript console except a few about preloading fonts.
Can't include previous version of stl/gcode as it was overwritten on the source during the edits.

cp2004, 2022-11-27:
Please test in safe mode before reporting your bug. If you aren't able to do that due to time constraints, as it says in the issue form please wait until you *are* able to do so before reporting your bug. There's time constraints on how long we will chase after information too.

cp2004, 2022-11-27:
Sounds like a duplicate of this issue #4200?

frollard, 2022-11-27:
> Sounds like a duplicate of this issue #4200?
Duplicate.
Almost certainly. My search-fu failed as I didn't see that one.

closed 2022-11-27 by frollard