OctoPrint/OctoPrint #4690: Newline characters missing from installing plugin log

TennisRunner, 2022-12-03:
### The problem

When I copy the output log from the installing plugin popup modal and paste it into vs code.

It does not contain any newline characters so the output is a mess to read.

For example:

[https://i.imgur.com/9biOVuh.png](https://i.imgur.com/9biOVuh.png)

> Installing plugin from uploaded file.../usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpqmjokxxp.zip --no-cache-dirProcessing /tmp/tmpqmjokxxp.zipERROR: Could not find a version that satisfies the requirement python-broadlink (from psucontrol-broadlink-subplugin)ERROR: No Installing plugin from uploaded file.../usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpqmjokxxp.zip --no-cache-dirProcessing /tmp/tmpqmjokxxp.zipERROR: Could not find a version that satisfies the requirement python-broadlink (from psucontrol-broadlink-subplugin)ERROR: No matching distribution found for python-broadlinkError!Could not parse output from pip, see plugin_pluginmanager_console.log for generated outputmatching distribution found for python-broadlinkError!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Should be:

> Installing plugin from uploaded file...
> /usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpqmjokxxp.zip --no-cache-dir
> Processing /tmp/tmpqmjokxxp.zip
> ERROR: Could not find a version that satisfies the requirement python-broadlink (from psucontrol-broadlink-subplugin)
> ERROR: No matching distribution found for python-broadlink
> Error!
> Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

Android 

### Browser and version of browser, operating system running browser

Firefox 107.0.1 (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2022-12-11:
I am wondering if this is a browser bug/incompatibility - I am not able to reproduce it on Edge (108.0.1462.46 (Official build) (64-bit)). I will fire up BrowserStack to see if I can try your Firefox (107).

There must be some newlines somewhere - the browser is rendering the log correctly after all.

foosel, 2023-01-24:
Not a browser incompatibility, a missing `\n` suffix. Affects both plugin management and software update.

foosel, 2023-01-24:
Fixed in 44b085abf6fafe361bef77019c41fa3f3fba854f (forgot to tag this ticket), will be part of 1.9.0

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel