OctoPrint/OctoPrint #4691: Unable to delete files

lettore, 2022-12-11:
### The problem

Nearly every time I update Octoprint or some 3rd party plugins it happens that I'm unable to delete files or files are not processed and scanned when uploaded.
I perfectly understand that it's a problem of a plugin that it's somewhat incompatible with the changes in Octoprint, but this is unacceptable at design level.
A plugin should not be allowed to break basic functionality of Octoprint, like deleting files.
Octoprint should report why is not able to accomplish the requested action and point to which plugin is the culprit.
It will be nice that, as it happens to Nextcloud for example, at every upgrade there's a list of plugin that it's not compatible with the new version, so a user can decide if upgrade or not.
It's a nightmare to diagnose every time which plugin can cause the problem, disabling one plugin at time and rebooting Octoprint.
Besides this, I'm wondering why wth minor upgrades Octoprint brings so many changes to break plugin functionality.
Definitively Octoprint needs a more robust way to handle how plugins are allowed to interact with the main program.
In this case the culprit is Continous Print 2.2.0, I already opened a bug on their github.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

Ubuntu

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20221211120327.zip](https://github.com/OctoPrint/OctoPrint/files/10201988/octoprint-systeminfo-20221211120327.zip)

cp2004, 2022-12-11:
We try hard not to break compatibility with plugins as much as is reasonably possible, and where breaking changes must happen we try and communicate as well as we can that plugins must be updated with warnings & notices for upcoming changes.

I can't tell from your issue here if this is related to some specific change in OctoPrint or not, I don't know what the plugin does that breaks deleting files to offer any insight into why it broke. Rest assured there were no intentional breaking changes for plugins & file deletions.

> I perfectly understand that it's a problem of a plugin that it's somewhat incompatible with the changes in Octoprint, but this is unacceptable at design level.

The system is rather robust already given plugin authors can write literally *any* Python or JavaScript code and OctoPrint has to try and handle it. It would be a much larger breaking change to start clamping down on what can/can't be done to make it acceptable for you. Like I already said, we didn't design any breaking changes, and usually like to know what goes wrong so we can avoid issues in the future.

> It will be nice that, as it happens to Nextcloud for example, at every upgrade there's a list of plugin that it's not compatible with the new version, so a user can decide if upgrade or not.

There are simply not enough hours in the day to test the 360+ plugins against every release of OctoPrint for one full time developer with the rest of us helping out in whatever spare time we have (currently not very much for me unfortunately). We have had ideas of created automated testing before, so if you have any experience with this please contribute. We run on average a month-long 'release candidate' cycle where we rely on the community to test OctoPrint with their plugins and report back on compatibility issues so they can be fixed - and yes this definitely isn't watertight.

> It's a nightmare to diagnose every time which plugin can cause the problem, disabling one plugin at time and rebooting Octoprint.

I'd recommend a binary search method - disable half, test, disable half of the two halves you have, repeat and you can narrow it down significantly quicker.

> Besides this, I'm wondering why wth minor upgrades Octoprint brings so many changes to break plugin functionality.

I think I've addressed this one above, we avoid it as much as possible and rely on the community to test & report things that get missed out accidentally.

lettore, 2022-12-11:
> I don't know what the plugin does that breaks deleting files to offer any insight into why it broke.

I opened a Github issue for the plugin Continous Print 2.2.0 [opened issue on Github](https://github.com/smartin015/continuousprint/issues/168)

I totally understand that being an open-source project the time is limited to who freely contributes to it, but I'm just asking a method that can reliably make the system work. 
It would be nice to put the compatibility test on the plugin developers. A plugin developer should have a month to test the plugin with the new version, marking it as compatible, non-compatible, or untested. 
Obviusly not every plugin developers will be available to test any new version, but this will narrows down the diagnosis to those that are untested or incompatible.

jneilliii, 2022-12-11:
> It would be nice to put the compatibility test on the plugin developers.

Not all plugin developers are actively working on their plugins and/or maintaining them as well as we would like. It seems from your posted issue that there already is a solution to your problem and you got that answer in what seems to be a very fast response time, it was just the time it took for you to figure out which plugin was causing the issue that is frustrating. 

> I totally understand that being an open-source project the time is limited to who freely contributes to it, but I'm just asking a method that can reliably make the system work.

You can help make this happen by participating in the month long RC phase and reporting issues that you find. The nice thing about finding these issues during that period is that the issues get fixed prior to final release (typically) and if it's enough of an impact to your workflow you can just switch back to stable release to revert back.

lettore, 2022-12-11:
From a user point of view wasting time trying to figure out where is the problem in RC phase or on the final release is the same.
It's really more simple for a developer to have an Octoprint installation with just the plugin that it's developing and see if it's behaving well or not.
I'm talking about Plugins that are officially listed, so it should be maintained the bare minimum to work with current releases.
I know the world is not perfect, this applies also to Octoprint, I'm just sharing an idea to develop a more clean update workflow that doesn't break working systems.
As user I think for me it will be safer to not install any updates if not required, because in the last 3 updates I had to disable some plugins for make the system working again.
As the current bug is already addressed to the plugin developer feel free to close this issue.
Thank you

cp2004, 2022-12-11:
> From a user point of view wasting time trying to figure out where is the problem in RC phase or on the final release is the same.

This is not true. If you update to an RC build, you are expecting it to be unstable and report issues with it. I wouldn't install it if I had important things to use OctoPrint for coming up, I would wait until I had time to test it. A few thousand users testing a release candidate, finding bugs and then us fixing them is *way way* better than releasing to a few hundred thousand users and then getting an endless stream of complaints like this one.

> It's really more simple for a developer to have an Octoprint installation with just the plugin that it's developing and see if it's behaving well or not.

I think you underestimate the amount of effort it takes to develop test & release an OctoPrint plugin. Hardly any developers get any compensation for doing so either. One person testing a plugin is not efficient nor feasible for many compared to the collective community effort to report bugs. If this were to be a requirement of having a plugin listed then you would be complaining that there were so few plugins in the repository instead.

I test all my plugins when I develop them and against new OctoPrint releases. Yet still, there are bugs in my plugins that users report and I fix. It is *impossible* for me to do all the testing myself and tell you that there will not be any bugs. I don't use my plugins the same way other people do. They might have two other plugins that conflict. The combinations are ridiculously high to expect one person to test in their spare time.

This conversation will not get anywhere without you becoming familiar with the development & testing process first, to understand the practicality of your ideas.

closed 2022-12-11 by cp2004

lettore, 2022-12-11:
I'm not judging you if you are familiar or not with the development and testing process, so it will be fair that you also don't judge the abilities of others as is not a point of this issue, and you don't know me personally. Have fun with your development and testing abilities, for sure I will not spend my time trying to fix something that is broken at the beginning for someone that judges others.