OctoPrint/OctoPrint #4692: Stuck at 10c below target temp

theyongdai, 2022-12-14:
### The problem

I have a Creality cr-10s, I get it is not the best printer in the world. It does get the job done. However, it seems to stuck at 10c below target temp. Anyone know what's going on here?

[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Creality cr-10s

### Browser and version of browser, operating system running browser

firefox

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Any g-code

[octoprint-systeminfo-20221213225529.zip](https://github.com/OctoPrint/OctoPrint/files/10224149/octoprint-systeminfo-20221213225529.zip)
 
[image]

cp2004, 2022-12-14:
This is not a bug with OctoPrint, please use the community for support - far more people can help there than the few who monitor bug reports.

---

This issue tracker is for bugs & feature requests for OctoPrint. For help and support in using OctoPrint, plugins, your printer and your general setup please get in touch on the [community forums](https://community.octoprint.org) or the [discord server](https://discord.octoprint.org) - helpful people are around that can answer your questions.

closed 2022-12-14 by cp2004

theyongdai, 2023-01-02:
Following up to my thread. I manage to dust out and bring online Simplify3D v4. It printed without any problems via usb. If anyone in the same boat, consider to give this a try. I'm very disappointed with the new octoprint updates, there seems to be problems with the target temperature set in octoprint. The development team seems to not acknowledging it.

[image]

[image]

jneilliii, 2023-01-02:
Can't acknowledge anything we can't reproduce.  I've been on the latest updates since rc phase and have never had this happen. You have undervoltage warnings in your log, and serial logging was not enabled during the issue and it appears safe mode wasn't started during the timeframe of the included octoprint.log in system bundle. Please enable serial logging, restart in safe mode and reproduce the issue. Also please include example gcode file used during testing.

cp2004, 2023-01-02:
Did you make any posts in the community forums or discord? We are more than happy to help but there are *far* more people who look at posts there and are able to help, along with troubleshooting being easier in the community.