OctoPrint/OctoPrint #4693: [Request] Please allow to rename jobs (gcode files) and "star" those to keep.

AndKe, 2022-12-15:
### Is your feature request related to a problem? Please describe.

[image]

### Describe the solution you'd like

When working with FreeCad and/or other tools, exported items may get different names.
Then - the objects might be duplicated in the slicer - so even if a job is automatically named: "Mount for xyz"  - it can be desired to save it as "3x Mount for xyz"   or something that describes the material and surface: "Mount for xyz - PLA on PEI"

Please allow renaming files. - good jobs may be worth saving.

Also: the files have properties like upload date, printed date and so on. It would be very nice to have a simple bool attribute "starred"  - then the file list could be sorted by "starred" (worth keeping jobs)   - with good names.

jneilliii, 2022-12-15:
Try pressing that little cut/scissor button and putting in a new name?

foosel, 2023-01-16:
As stated by @jneilliii, renaming (and moving) is already possible via the scissors icon. Which makes this a request to be able to star dear-to-heart print jobs. And that makes it a duplicate of #3433

closed 2023-01-16 by foosel