OctoPrint/OctoPrint #4698: [Request] Improve integration with Edge's tab sleeping

cp2004, 2022-12-19:
### Is your feature request related to a problem? Please describe.

Currently, when a tab is not being used for a little while (5 mins on my laptop), Edge puts the tab to 'sleep' which means it stops it doing background things.

The websocket seems to stay connected but messages/UI updates are not processed by OctoPrint until the tab wakes up.

This leads to a period of time where the UI is trying to 'catch up' and processes all the messages, and it can take a while to do that if the tab has been sleeping for a little while - it's always easier/quicker to reload the page.

I would like to investigate if there is anything OctoPrint can do to effectively skip to the newest websocket messages, or disconnect & reconnect the socket when it is sleeping so the messages do not build up in the backlog. Perhaps there is a browser event OctoPrint could listen to. I know it is quite browser specific, so if the solution is complex then we could leave it. But if it is only a couple of additional lines, I would like to see how we could make it work.

### Describe the solution you'd like

already done

### Describe alternatives you've considered

Leaving it alone

### Additional context

Might try and record a video demonstrating the problem at some point.

foosel, 2023-01-16:
> I would like to investigate

Took the liberty to assign it to you :)

seanmcne, 2023-02-14:
I've noticed the same strange issues - the way the web interface works in edge is it seems to 'pause' when it sleeps and then tries to fast forward through everything so it lags the whole tab down as it quickly speeds up the time and tries to render the video, etc.  It's kind of a mess for maybe 30 seconds or so - I suppose one workaround is to force edge to not sleep tabs from the website?

ivebeenlinuxed, 2025-03-03:
Hello! FWIW I had a quick look at some documentation on this. It seems that there isn't an API to directly tell whether a tab has gone to sleep (Here is a crude attempt at a workaround: https://gist.github.com/furf/b57ce352b506660624b6328c71c3788b)

The gist did mention the Page Visibility API however (https://developer.mozilla.org/en-US/docs/Web/API/Page_Visibility_API) which may help. Rather than disconnecting the websocket on sleeping, it instead could work by disconnecting when a tab went invisible (or maybe after a length of time invisible to stop rapid/annoying disconnects).

caniuse suggests there is good browser support for this: https://caniuse.com/?search=page%20visibility

ivebeenlinuxed, 2025-03-04:
I have included a POC here:

https://github.com/ivebeenlinuxed/OctoPrint/commit/5c6f3db7c4e54285c5cd6c824983d6d5a8521fa3

Of course, this raises a number of questions about the state of the UI once the reconnection happens...

cp2004, 2025-03-04:
Thanks @ivebeenlinuxed - I think Page Visibility API would be the way forward for this too. +However_ I think I bumped into the same question - there are assumptions made in both core & plugins that assume the Websocket has remained connected & processes all messages. For example, processing one-time events - if this makes the UI state out of sync with the server, and you have a problem. Difficult one to solve with OctoPrint's architecture I think.