OctoPrint/OctoPrint #4701: Octoprint server login issue

Malbeex, 2022-12-22:
### The problem

I changed the password for my octoprint login and now it won't let me log in again. Keeps giving me error messages. It won't accept the old or new password and there is no option for changing or resetting it unless I am logged in.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I can't log in to the server 

### Version of OctoPrint

1.3.6

### Operating system running OctoPrint

Windows

### Printer model & used firmware incl. version

Makergear M3-ID running octopi 0.13.0

### Browser and version of browser, operating system running browser

Google Chrome on Windows 11

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]

Malbeex, 2022-12-22:
I use the online platform I do not have octopint installed on my pc yet so I cant access the settings tab unless I log in

jneilliii, 2022-12-22:
SSH to your pi and run the command `~/oprint/bin/octoprint systeminfo .` (the dot is important). then use winscp or filezilla to grab the file it generates from the pi. You can reset the password from the command line to. https://community.octoprint.org/t/i-forgot-my-octoprint-password-how-can-i-reset-it/215?u=jneilliii

cp2004, 2022-12-23:
> Version of OctoPrint
> 1.3.6
> Printer model & used firmware incl. version
> Makergear M3-ID running octopi 0.13.0

You are running a version of OctoPrint provided by Makergear on their printer. Your first port of call for support should be the manufacturer since they know what they customised on their install, we can't support these installs here. OctoPrint 1.3.6 is over 5 years old now and would be no longer supported regardless.

If you are able to install a modern version of (mainstream) OctoPrint, then for help and support in using OctoPrint, plugins, and your general setup please get in touch on the [community forums](https://community.octoprint.org) or the [discord server](https://discord.octoprint.org) - helpful people are around that can answer your questions. 

[This FAQ post](https://community.octoprint.org/t/i-forgot-my-octoprint-password-how-can-i-reset-it/215) details the steps to reset your password on a normal OctoPrint install, sadly I can't say if it would apply to Makergear's version of OctoPrint.

closed 2022-12-23 by cp2004