OctoPrint/OctoPrint #4706: Temperature offset lost upon Octoprint restart

elpeter25, 2023-01-06:
### The problem

Temperature offset lost upon Octoprint restart

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

Ubuntu

### Printer model & used firmware incl. version

Prusa MK3 latest FW (3.11)

### Browser and version of browser, operating system running browser

Firefox 108.0.1 (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230106112153.zip](https://github.com/OctoPrint/OctoPrint/files/10359353/octoprint-systeminfo-20230106112153.zip)

https://user-images.githubusercontent.com/37652763/210994181-d64ad4a3-60cd-404b-b11a-365e72d30093.mp4

cp2004, 2023-01-06:
This is, as far as I can tell, expected behaviour. The offsets aren't stored anywhere, they are temporary.

elpeter25, 2023-01-06:
wops! 
Then can I ask for that feature? I have printers with Nozzle X which requires a offset temperature about at least +5º... and I want it to always be like that.

cp2004, 2023-01-15:
I've converted this to a feature request. 

It could be something that is added to the printer profile (a 'default temperature offset'). So it still shows up & is editable (any changes stored temporarily) in the UI, just with a different starting value.

elpeter25, 2023-01-17:
> I've converted this to a feature request.
> 
> It could be something that is added to the printer profile (a 'default temperature offset'). So it still shows up & is editable (any changes stored temporarily) in the UI, just with a different starting value.

Thanks!