OctoPrint/OctoPrint #4707: Documentation for heater parameter of set_temperature method in PrinterInterface is missing "chamber"

CMR-DEV, 2023-01-06:
### The problem

This is more a gap in documentation than a functional bug:
`PrinterInterface`'s `valid_heater_regex` in [`src/octoprint/printer/__init__.py:49`](https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/printer/__init__.py#L49) matches strings like `"bed"`, `"chamber"`, `"tool0"`, `"tool1"` and so on. It is used for validating the `heater` parameter in [set_temperature](https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/printer/standard.py#L569). However, `"chamber"` is not mentioned in its interface documentation: [`src/octoprint/printer/__init__.py:270`](https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/printer/__init__.py#L257)

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

The issue is not related to any running code

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

OctoPi

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2023-01-06:
Would you like to send a PR to fix it?

CMR-DEV, 2023-01-06:
I will

foosel, 2025-02-05:
Totally forgot to close this, it has been fixed since 1.10.0

closed 2025-02-05 by foosel