OctoPrint/OctoPrint #4711: Stream & Snapshot URL lost when pulling in webcam plugin changes

cp2004, 2023-01-21:
### The problem

I updated my local copy of maintenance, with the new changes from #4628. When I restarted OctoPrint, the stream URL was empty in the 'Classic webcam' settings.

These URLs should be migrated out of the config. They are still in config.yaml under `webcam`, but haven't been moved to the classic webcam plugin.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.9.0.dev291+ge335055a9

### Operating system running OctoPrint

Windows 11

### Browser and version of browser, operating system running browser

Edge

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

I expect my systeminfo bundle is probably full of crap from various developments, but I'll attach it anyway...

[octoprint-systeminfo-20230121235412.zip](https://github.com/OctoPrint/OctoPrint/files/10473387/octoprint-systeminfo-20230121235412.zip)

cp2004, 2023-01-21:
False alarm, I must have tested the PR at an earlier stage (& since forgotten) but it meant that the settings migration didn't fire. All is well.

closed 2023-01-21 by cp2004

crysxd, 2023-01-22:
Was already shocked :D I thought I tested that all really well