OctoPrint/OctoPrint #4712: Authorization flow not triggered after sign in

crysxd, 2023-01-22:
### The problem

When a 3rd party client launches [the authorization flow](https://docs.octoprint.org/en/master/bundledplugins/appkeys.html#sec-bundledplugins-appkeys-workflow) but the user is not yet signed in in the browser the 3rd party client uses the sign in is shown. After the sign in is completed the user is forwarded to the normal UI, not the authorization flow.

To reproduce:

- Start the authorization flow and extract the `authUrl`
- Open a Incognito window in Chrome/Edge/Safari
- Open the link from `authUrl`
- Sign in is shown
- Complete sign in
- Expected behavior: Authorization request is shown
- Actual behavior: Main UI is shown

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

/

### Browser and version of browser, operating system running browser

Safari, Chrome, Edge (all latest)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230122065339.zip](https://github.com/OctoPrint/OctoPrint/files/10473886/octoprint-systeminfo-20230122065339.zip)
[jslogs.log](https://github.com/OctoPrint/OctoPrint/files/10473887/jslogs.log)
https://user-images.githubusercontent.com/5859228/213904524-6273ef63-4c14-48c0-91fb-954e08ffad2e.mov

foosel, 2023-01-23:
On second look, the solution is actually to extend `allowed_paths` here:

https://github.com/OctoPrint/OctoPrint/blob/94408fd2dd3407666ba5c8f279eb4dcf6e75b2a2/src/octoprint/server/views.py#L184

The problem now is, I'm not sure if this should include all endpoints registered by plugins, or only a subset.

foosel, 2023-01-23:
Great point, that's what we'll do! And just so that if push comes to shove, there's an option to manually intervene, I'll add a setting that's `config.yaml` only (so no UI for it, no API, nothing)

crysxd, 2023-01-23:
Sounds like a great solution. Do you want me to make a PR? Might also be just quicker if you do it yourself, though :D Up to you :)

foosel, 2023-01-23:
Already on it, but thanks :)

foosel, 2023-01-23:
Should be fixed by the above commit, will be part of 1.9.0.

Tbh, kinda surprised it took until now for this to be found, this has been broken since 1.8.0 was released back in May of last year. But I guess people usually are already logged in before they use the appkey workflow with some third party client 😅

crysxd, 2023-01-23:
Well, it's also not a big deal if you are not logged in. The authorization is directly shown in the top left, so the user can still achieve the goal. Not quite as comfortable, but could be worse.

commit c7dee8d6 (2023-01-23): 🐛 Allow redirect to appkeys from login dialog

Closes #4712
closed 2023-05-23 by foosel