OctoPrint/OctoPrint #4724: Height in `.metadata.json` is always off by ~+0.2mm

CoryCharlton, 2023-02-03:
### The problem

The `height` value in the `analysis:dimensions:height` field found in the `.metadata.json` file is always off by ~+0.2mm. 

This value matches the  `analysis:printingArea:maxZ` field. My custom end GCODE from Cura includes the following line:

`G1 E-2 Z0.2 F2400 ;Retract and raise Z`

So I assume `analysis:printingArea:maxZ` is correct but since no extrusion occurs at this height it should not be included in the dimensions (unless of course they represent something other than the "printed" dimensions).

The GCODE viewer tab handles this correctly.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.8.6

### Operating system running OctoPrint

OctoPi 1.0.0

### Printer model & used firmware incl. version

Creality Ender 3 V2 - Ender3 V2/S1 Professional Firmware 20221222

### Browser and version of browser, operating system running browser

N/A

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230203103943.zip](https://github.com/OctoPrint/OctoPrint/files/10581792/octoprint-systeminfo-20230203103943.zip)

The attached GCODE is 19.0mm high but the metadata analysis shows it as 19.200000762939453mm.

[CE3E3V2_top-insert -no-reel.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/10581876/CE3E3V2_top-insert.-no-reel.gcode.zip)
[metadata.json.zip](https://github.com/OctoPrint/OctoPrint/files/10581879/metadata.json.zip)

[image]

[image]

jneilliii, 2023-02-03:
not sure if it would matter, but try changing your command to use `G0` instead of `G1` as that is a travel move command.

CoryCharlton, 2023-02-03:
> not sure if it would matter, but try changing your command to use `G0` instead of `G1` as that is a travel move command.

Thanks for the suggestion. It looks like analysis doesn't occur while a print is running so I'll report back the results from this file in ~4 hours: 

[CE3E3V2_top-insert -no-reel_Test_4724_with_G0.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/10582061/CE3E3V2_top-insert.-no-reel_Test_4724_with_G0.gcode.zip)

cp2004, 2023-02-03:
I've just uploaded your file (from the initial report) to my checkout of maintenance (soon to be OctoPrint 1.9) and can't reproduce the issue, it shows 19.0 for me:

[image]

Your file (or at least the associated metadata that you have uploaded) is being processed by DisplayLayerProgress - not sure if that will make a difference or not. I'll check out what a stable 1.8.6 does if I get a chance.

CoryCharlton, 2023-02-03:
@cp2004 just to confirm, the value in the .metadata.json file is also 19.0?

I initially started with DisplayLayerProgress ([#272](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/272#issue-1568923135)) but my quick review of the code suggests that it only reads `analysis:dimensions:height` and does not write to it.

I just checked out the maintenance branch and this [commit](https://github.com/OctoPrint/OctoPrint/commit/4c86459c1bb2cd72152ffad37bf5ad05b9159276#diff-9cbdfbf0aa92e7591657e594e39cbb0bad0edb8ef9640ea32d887ca13f69a870L466) (or something else related to these changes) may be the difference you are seeing (assuming the metadata is correct in 1.9).

CoryCharlton, 2023-02-04:
> > not sure if it would matter, but try changing your command to use `G0` instead of `G1` as that is a travel move command.
> 
> Thanks for the suggestion. It looks like analysis doesn't occur while a print is running so I'll report back the results from this file in ~4 hours:
> 
> [CE3E3V2_top-insert -no-reel_Test_4724_with_G0.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/10582061/CE3E3V2_top-insert.-no-reel_Test_4724_with_G0.gcode.zip)
> 
> **Edit:**
> 
> From:
> 
> https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/util/gcodeInterpreter.py#L425
> 
> It looks like `G0` and `G1` will be treated the same. What I'm not clear on is why `e` would be positive if that GCODE line is the one triggering this issue:
> 
> https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/util/gcodeInterpreter.py#L467

Confirmed that G0 or G1 makes no difference.

Also confirmed that DisplayLayerProgress is not causing the issue by uninstalling it.

jneilliii, 2023-02-04:
I can confirm this is resolved in 1.9.0 and reproducible in 1.8.6

1.9.0 (1.9.0.dev161+gfd811acf8)
[image]

1.8.6
[image]

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel