OctoPrint/OctoPrint #4725: [Request]

rp10007, 2023-02-05:
### Is your feature request related to a problem? Please describe.

When I open the Octoprint UI and the printer is off (which is almost always), the communications section displays an alert: 

"No serial port found, are you sure your printer is physically connected and supported? Try refreshing and if that doesn't help please see [the FAQ](https://faq.octoprint.org/no-candidates)."

Which is fine in theory, but in practice my printer is off because I turned it off using PSU control or using the post-print timeout. I know it's off, and I'm going to turn it on using the PSU control as soon as I finish scanning the UI for alerts that might be important. This isn't a big deal, but it's still a distraction every time. 

### Describe the solution you'd like

Ideally I'd like some kind of protocol between plugins and Octoprint so the plugin can say "I'm turning the printer off now, don't worry" or "I'm turning the printer on now, you should start caring about enumeration again." And then Octoprint would either not display an alert during the known-to-be-off time or display a simpler "The printer was turned off" alert (perhaps with a string provided by the plugin). Octoprint might also forego checking for a serial port during known-to-be-off times (although in the fanciest version of this protocol, you might want an alert when the printer stays on when it's supposed to be off). 

### Describe alternatives you've considered

Perhaps rewrite the existing alert to say "Is your printer turned on?" The current text is useful for a first-time user or someone with a newly-attached printer, but a hindrance (imo) for the majority of users once their setup is properly configured. (Or maybe a choice of alert texts? although that seems like a rabbit hole.)

jneilliii, 2023-02-05:
You know you can disable the warning? Under Serial Connection settings...

[image]

rp10007, 2023-02-05:
Sorry about that. Need to work on my reading skills.

closed 2023-02-05 by rp10007