OctoPrint/OctoPrint #4733: [Request]

KoesterJannik, 2023-02-13:
### Is your feature request related to a problem? Please describe.

I want error messages from the rest api response. It is very hard to determine whats wrong in an 400 error for example.

### Describe the solution you'd like

Return a valid message as json.

### Additional context

It is very hard to trial and error the requests in order to figure it out...

cp2004, 2023-02-13:
Which API route are you using? Some of them do show errors, eg

https://github.com/OctoPrint/OctoPrint/blob/a1d95dd44b2c78e0bd609c8ef4f7873fa66d07d4/src/octoprint/server/api/job.py#L17

KoesterJannik, 2023-02-13:
Im using /api/file routes. i just get plain 400 status codes with no message

closed 2023-02-13 by KoesterJannik

cp2004, 2023-02-13:
Did you mean to close this?