OctoPrint/OctoPrint #4734: Print quality via Octoprint significantly worse than directly from SD-Card - Ender-5 S1

TomMuc1, 2023-02-16:
### The problem

print quality via octoprint significantly worse than directly from sd-card

with exactly the same g-code file the print result is very clearly worse than when the printer is controlled with octoprint via usb-c. this is not only the case with the attached example file but with every file.

-> raspberry pi 4 model b rev 1.2 2gb
-> octopi installation via raspberry imager
-> no time lapse or similar active

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

follow up

### Version of OctoPrint

OctoPrint 1.8.6 Python 3.7.3 OctoPi 0.18.0

### Operating system running OctoPrint

raspbian buster

### Printer model & used firmware incl. version

creality ender-5 s1

### Browser and version of browser, operating system running browser

safari macos 13.2.1

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230216225509.zip](https://github.com/OctoPrint/OctoPrint/files/10761282/octoprint-systeminfo-20230216225509.zip)
[E5 S1-rabbit.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/10761289/E5.S1-rabbit.gcode.zip)
[image]

wakeupcairo, 2023-07-22:
any update abut this bug?

Sveeeeeen, 2023-09-01:
Facing the same issue, but my Raspi 4 B has 8GB of ram.

foosel, 2023-11-28:
If you are still seeing this against current versions, then please enable [`serial.log`](https://faq.octoprint.org/serial-log), reproduce the issue, create another system info bundle and attach it here. Stuttering and quality issues can be rooted in communication issues, and sadly your provided logs don't contain any serial output to check for issues like that.

closed 2024-04-11 by foosel