OctoPrint/OctoPrint #4735: Firmware & Protocol settings do not save

hagan66, 2023-02-17:
### The problem

I have a problem with my Prusa MK2.5s getting M112 and firmware crash after MMU load error. I wanted to see if these settings would help, as I read somewhere. (see attached screenshot)

But the boxes do not stay checked even after saving. 
[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

Octopi 0.18.0

### Printer model & used firmware incl. version

Prusa MK2.5S fw 3.12.1

### Browser and version of browser, operating system running browser

Safari 16.3

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230217101233.zip](https://github.com/OctoPrint/OctoPrint/files/10765636/octoprint-systeminfo-20230217101233.zip)

jneilliii, 2023-02-17:
Did you happen to uncheck this box too for enable firmware detection?

[image]

hagan66, 2023-02-17:
Yes I had to uncheck that one for the other options to become visible.

jneilliii, 2023-02-17:
Confirmed, I'm able to reproduce this issue in OctoPrint 1.8.6 and current maintenance branch (future 1.9.0 version).

foosel, 2023-02-20:
Confirmed as well. For some reason the server side API endpoint never tried to save these two settings. The only idea I have is that there was some merge issue a long time ago that caused this, but who knows.

Fix ready for 1.9.0

commit f02ce519 (2023-02-20): 🐛 Fix two settings not saving

Closes #4735
closed 2023-05-23 by foosel