OctoPrint/OctoPrint #4737: Bug?? Octoprint loses connection to marlin when M600 is triggered?

Voxel3DPrinting, 2023-02-20:
### The problem

Hi All,

I recently hooked my Xinkebot Orca 3 IDEX printer up to octoprint. I am really excited about the program and it works great. However, it now does give me a headache when M600 is triggered via the filament sensor;

I have installed 2 BTT Smart filament sensors connected to the BTT octopus V1.1 mainboard. When printing from the sd card on the printer, they both work perfectly.
When printing from octoprint and sending "M600" via the terminal, it also works fine.

However, when the M600 is triggered by the filament sensors, and the current print is running via octoprint, it pauses, moves the printhead to the park position, and then loses connection to marlin.. (Error; "Too many consecutive timeouts, printer still connected and alive?") Because of this I cannot resume the print anymore..

I have made sure that both #define HOST_KEEPALIVE_FEATURE" and "#define HOST_ACTION_COMMANDS" are enabled in marlin.

I really do not know what could cause this issue... Any help?

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.8.6 

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Mainboard; BTT octopus V1.1 / Screen; BTT TFT35 V2.0 / Raspberry Pi 4 model B 4gb / Marlin version 2.1.2

### Browser and version of browser, operating system running browser

Microsoft edge and Chrome tested

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

System file bundle;
[octoprint-systeminfo-20230220001319.zip](https://github.com/OctoPrint/OctoPrint/files/10782585/octoprint-systeminfo-20230220001319.zip)

Video and messages;
https://user-images.githubusercontent.com/73597964/220065475-db37fcaa-5311-4605-b908-e7e9d0ef5b01.mp4

foosel, 2023-02-20:
Could indeed be a bug:

```
Recv: ok
Recv: //action:paused
Printer signalled that it paused, switching state...
Changing monitoring state from "Pausing" to "Paused"
Recv:  T:200.06 /200.00 B:50.02 /50.00 T0:200.06 /200.00 T1:20.95 /0.00 @:32 B@:33 @0:32 @1:0
Recv: echo:busy: processing
Recv:  T:200.18 /200.00 B:50.01 /50.00 T0:200.18 /200.00 T1:20.95 /0.00 @:31 B@:34 @0:31 @1:0
Recv: echo:busy: processing
Recv:  T:200.32 /200.00 B:50.00 /50.00 T0:200.32 /200.00 T1:20.91 /0.00 @:28 B@:35 @0:28 @1:0
Recv: X:180.33 Y:390.00 Z:20.00 E:238.53 Count X:14426 Y:31200 Z:16060
Recv: echo:busy: processing
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M105
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M105
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Paused" to "Offline after error"
Connection closed, closing down monitor
```

This suggests your printer properly reported the paused state to OctoPrint, then started to report it was busy. Yet OctoPrint did run into a communication timeout. Will need to look into why that happened, as *normally* the `echo:busy processing` should tell OctoPrint "Hey, all is well, the printer is still alive but cannot receive any more commands right now, please hold your horses".

However, something else I noticed in your system info bundle:

```
2023-02-19 22:51:17,569 - octoprint.plugins.pi_support - WARNING - This Raspberry Pi is reporting problems that might lead to bad performance or errors caused by overheating or insufficient power.
!!! UNDERVOLTAGE REPORTED !!! Make sure that the power supply and power cable are capable of supplying enough voltage and current to your Pi.
```

You need to fix your power supply, your Pi has run into some undervoltage. You will also have received a warning about this in the UI. This is most likely not related to the behaviour observed here however.

Voxel3DPrinting, 2023-02-20:
Yeahh I saw, I have original powersupply, so ignored the message.

Upon further inspextion of the pi I saw that some Pins had a could-solder joint. I'll try fixing that later.

Is there any way I can help fixing the bug? or keep beibg updated?

foosel, 2023-02-20:
I'm not yet 100% sure that it is a bug, trying to wrap my head around it right now.

What I can tell you is that your firmware isn't reporting that it is supporting the busy protocol, albeit it obviously is supporting it. That shouldn't cause this as far as I can see.

What would help a lot here is to get some timing information. Please enable `serial.log` (Settings > Serial Connection > General > Serial Logging), reproduce the error, then upload a new system info bundle. That way I'll get a `serial.log` of the full printer communication, with timestamps, not just the non-timestamped excerpt that was included in your current one's `terminal.txt`.

Voxel3DPrinting, 2023-02-20:
When heating or standing stationary in a position, it is reporting "echo: busy". Is this what you mean?

I will make a new file with serial logging enabled when I get home. I suspect it will be in 4 to 5 hours. 

Thanks for replying so quickly and help me figure out this issue. I'll help where I can.

foosel, 2023-02-20:
> When heating or standing stationary in a position, it is reporting "echo: busy". Is this what you mean?

Yes, that is the so called busy protocol. Normally, the printer should also be reporting that as a capability in its capability report as `Cap:BUSY_PROTOCOL:1` but that's missing from the one I see in your shared `terminal.txt`:

```
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin 2.1.2 (Feb 19 2023 23:45:39) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Orca 3 EXTRUDER_COUNT:2 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:HOST_ACTION_COMMANDS:1
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:0
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:0
Recv: Cap:LFN_WRITE:0
Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:0
Recv: Cap:EXTENDED_M20:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: Cap:CONFIG_EXPORT:0
Recv: ok
```

By default, without firmware-side busy protocol OctoPrint will wait 30s for an `ok` from the printer before it considers there to be a timeout. If it sees an `busy` or `echo:busy` during this it will also continue to wait. With `Cap:BUSY_PROTOCOL:1` OctoPrint will tell the printer to send the `busy` messages in a specific interval (2s by default) and thus will be able to react to a timeout faster than 30s. But since your firmware doesn't tell OctoPrint that it actually supports setting this interval (no positive capability report) it never tries to tell the printer that and goes with the 30s default. My current guess is that your `busy` messages come less often than 30s and thus a timeout can happen, but that's impossible to say for sure without a timestamped `serial.log`. If they come more frequently, then something else is going on.

Voxel3DPrinting, 2023-02-20:
Ahh, I understand.

The interval set in marlin is 5 seconds. However, when M600 is triggered, I can see that octoprint is still receiving input from the printer (changing state to "paused", etcetera) also, there is around 10 seconds max between the M600 trigger at the machine, and the disconnection of octoprint. This is way less than the 30 seconds you mention.

As said I will enable logging, recreate the issue and send the new system file at the end of today. That will make it more clear I hope

Voxel3DPrinting, 2023-02-20:
[octoprint-systeminfo-20230220154555.zip](https://github.com/OctoPrint/OctoPrint/files/10785403/octoprint-systeminfo-20230220154555.zip)

Here you go!

foosel, 2023-02-20:
> I can see that octoprint is still receiving input from the printer (changing state to "paused", etcetera)

FYI, that's not input from the printer, those are status messages from OctoPrint. Stuff coming from the printer is prefixed with `Recv:`, stuff going to the printer is prefixed with `Send:`, everything else is status info or messages generated by OctoPrint.

Looking at the `serial.log`, what I don't understand is why the printer is sending an `//action:paused` trigger twice, and I'm wondering if that might be the reason for the timeout: 

```
2023-02-20 15:45:12,159 - Recv: //action:prompt_end
2023-02-20 15:45:12,162 - Recv: //action:prompt_begin FilamentRunout T0
2023-02-20 15:45:12,163 - Recv: //action:prompt_show
2023-02-20 15:45:12,166 - Recv: //action:paused filament_runout 0
2023-02-20 15:45:12,167 - Printer signalled that it paused, switching state...
2023-02-20 15:45:12,168 - Changing monitoring state from "Printing" to "Pausing"
2023-02-20 15:45:13,078 - Recv:  T:200.13 /200.00 B:50.02 /50.00 T0:200.13 /200.00 T1:17.24 /0.00 @:32 B@:54 @0:32 @1:0
2023-02-20 15:45:13,130 - Recv: //action:paused
2023-02-20 15:45:13,131 - Printer signalled that it paused, switching state...
2023-02-20 15:45:15,051 - Recv:  T:200.16 /200.00 B:50.04 /50.00 T0:200.16 /200.00 T1:17.28 /0.00 @:31 B@:53 @0:31 @1:0
2023-02-20 15:45:15,111 - Recv: echo:busy: processing
2023-02-20 15:45:17,051 - Recv:  T:200.23 /200.00 B:50.04 /50.00 T0:200.23 /200.00 T1:17.31 /0.00 @:30 B@:53 @0:30 @1:0
2023-02-20 15:45:17,111 - Recv: echo:busy: processing
2023-02-20 15:45:18,800 - Recv: X:159.22 Y:390.00 Z:20.00 E:5.88 Count X:12738 Y:31200 Z:16060
2023-02-20 15:45:19,055 - Recv:  T:200.47 /200.00 B:50.05 /50.00 T0:200.47 /200.00 T1:17.23 /0.00 @:26 B@:51 @0:26 @1:0
2023-02-20 15:45:19,111 - Recv: echo:busy: processing
2023-02-20 15:45:22,116 - Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2023-02-20 15:45:22,121 - Changing monitoring state from "Pausing" to "Paused"
```

But in any case I'll have to see if I can somehow reproduce this on the virtual printer, because it also looks like it is waiting for something between the states pausing and paused, which it shouldn't. That will take some time though.

Voxel3DPrinting, 2023-02-20:
Thanks, no problem. I can send you my exact firmware file/configuration files if it would be convenient for you??
Any indication on how long it will take? Couple days / weeks / months / years?

foosel, 2023-02-20:
Can't say currently. Considering that it will be tricky to reproduce (firmware config might be helpful indeed, compiled firmware won't help much without that printer ^^) and if it's a bug it's been in there for a while, I've decided against putting it into 1.9.0 (unless it happens to fall in front of my feet), and that currently has priority to get a first RC for out of the door.

Voxel3DPrinting, 2023-02-20:
[Configurations.zip](https://github.com/OctoPrint/OctoPrint/files/10787451/Configurations.zip)

Here you go.

Again, if you need any testing or assistance, feel free to contact me.

Voxel3DPrinting, 2023-02-23:
ITS WORKING!!!!

Hi, 11 iterations of Marlin further, I've managed to get it working. I do not fully understand why, but setting the "#define TX_BUFFER_SIZE 0" to 32, and changing the "communication timeout busy protocol support detected" from 3 to 15 fixed the issue for me.

I can now use my 2 smart filament sensors connected to the mainboard correctly with Octoprint

closed 2023-04-21 by Voxel3DPrinting