OctoPrint/OctoPrint #4744: 500 response when requesting `/api/currentuser` caused by Babel

crysxd, 2023-03-02:
### The problem

As discussed on Discord: https://discord.com/channels/704958479194128507/705047010641838211/1080750696430841856
And in the forum: https://community.octoprint.org/t/http-500-when-trying-to-connect-to-the-api/50252

```
2023-03-02 06:41:25,493 - octoprint - ERROR - Exception on /api/currentuser [GET]
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python3.10/site-packages/flask/app.py", line 1519, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python3.10/site-packages/flask/app.py", line 1515, in full_dispatch_request
    rv = self.preprocess_request()
  File "/usr/lib/python3.10/site-packages/flask/app.py", line 1857, in preprocess_request
    rv = self.ensure_sync(before_func)()
  File "/usr/lib/python3.10/site-packages/octoprint/server/__init__.py", line 1427, in before_request
    g.locale = self._get_locale()
  File "/usr/lib/python3.10/site-packages/octoprint/server/__init__.py", line 1352, in _get_locale
    return Locale.parse(request.accept_languages.best_match(LANGUAGES))
  File "/usr/lib/python3.10/site-packages/babel/core.py", line 317, in parse
    raise TypeError(f"Unexpected value for identifier: {identifier!r}")
TypeError: Unexpected value for identifier: None
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.6

### Operating system running OctoPrint

/

### Printer model & used firmware incl. version

/

### Browser and version of browser, operating system running browser

/

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230301234452.zip](https://github.com/OctoPrint/OctoPrint/files/10872077/octoprint-systeminfo-20230301234452.zip)

foosel, 2023-03-02:
Reason is most likely that we are blindly providing `Local.parse` with the result from `request.accept_languages.best_match`, and I guess that can be `None`. Not an issue so far, but Babel released 2.12 two days ago and apparently that no longer supports `None`. And since our version of `Flask-Babel` doesn't force an upper limit for `Babel`, and this was a minor version increase anyhow, this breaking issue was not expected at all and now here we are.

foosel, 2023-03-02:
https://github.com/python-babel/babel/commit/9b67367950e2ae95fdea585886a564f3ce8cdb5d is the commit that caused this behaviour change in Babel.

Note that the method signature even still says it accepts `None`. This is a breaking change in behaviour I certainly would not expect on a minor version upgrade 😕 Couldn't find it explicitly stated in the changelog either.

Guess I'll have to push 1.8.7 now 🙄

closed 2023-03-02 by foosel

Uro1, 2023-03-02:
> Note that the method signature even still says it accepts None. This is a breaking change in behaviour I certainly would not expect on a minor version upgrade 😕 Couldn't find it explicitly stated in the changelog either.

The `CHANGES.rst` of python-babel was bizarrely updated after the release (https://github.com/python-babel/babel/pull/978) documenting the change after it was highlighted to them as a breaking-change by way of this issue 2 days ago (the same day 2.12.1 was released) - https://github.com/python-babel/babel/issues/977

I would have thought any breaking-change would warrant a major version bump to highlight the change to the users of python-babel so those users can factor it in and update thier own code to handle the changes correctly.

NotReallyADeveloper, 2023-03-02:
As one who's running into this as well, I was happy to see a solution turned around quickly.  Upgrading from the CLI didn't work, and it wasn't showing up in Stable, but setting to Maintenance RC showed the release as being available.

Anyhow, after installing the new version, I'm happy to say the error went away (I was having difficulty with Home Assistant and Octoprint integration not being able to get the API keys.

Thank you for the fix and the great software, too!

cp2004, 2023-03-02:
> it wasn't showing up in Stable, but setting to Maintenance RC showed the release as being available.

That's likely to do with the caching of the update checks - it can take up to 24 hours to come up 'normally', and then when you switch release channels a refresh is done instantly.

NotReallyADeveloper, 2023-03-02:
Thanks.  Not sure I'm completely out of the woods; while HA can now register the plugin, none of the devices are responding.  I'll create a new issue.

commit 59054cc1 (2023-03-02): 🥅 Don't send `None` to `Locale.parse`

The latest version of Babel no longer supports this
but raises an error, which in turn leads to a
500 Server Error response here.

To fix this, we set a default for best_match.

Closes #4744